alexforencich/verilog-ethernet #92: VLAN support

thienanvu, 2021-09-14:
Hi,

Thank you for sharing your Ethernet/UDP/IP core. Just wonder if VLAN in the MAC layer is supported?

Thanks

alexforencich, 2021-09-14:
VLAN tags are not currently supported.

thienanvu, 2021-09-15:
Thank you Alex. Is it something feasible to add to your cores? I am thinking of trying you UDP/IP block with the Xilinx Ethernet Subsystem that support VLAN.

alexforencich, 2021-09-15:
Well, the MACs in here do not care about VLAN tags.  They simply pass the whole frame through, VLAN tag and all.  So from that standpoint it doesn't matter if you use the MACs here or the Xilinx MAC.  The problem is the rest of the stack - there are no signals to carry the VLAN tag, so that would have to be added to basically everything.  And the Ethernet frame handling modules would have to be updated to insert/remove the tag.

thienanvu, 2021-09-15:
Thank you for your clarification. Are the Ethernet frame handling in eth_axis_rx/tx modules?