alexforencich/verilog-ethernet #94: Sending large UDP packets

IvanSavelyev, 2021-09-21:
Thank you for the wonderful IP modules.
I want to ask a question about the examples given in the repository.
I have successfully launched a project for DE2-115, and I am trying to send and receive back a packet with data containing more than 1500 bytes in the packet
It seems that the UDP protocol is capable of transmitting 64 kB of data, but in the example, as I understand it, only one packet is returned.
The question is whether this IP is capable of transmitting a packet of more than 1500 bytes, if so, how can this problem be solved.

alexforencich, 2021-09-21:
Fragmentation is not supported, so you're limited to whatever you can send in a single MTU.  But the MTU is not limited to 1500 bytes, you can certainly change the FIFO sizes and run with 9 KB jumbo frames if you want to.

IvanSavelyev, 2021-09-23:
It looks like I'll have to go into the network card settings.
Thanks for the reply.

closed 2021-09-24 by IvanSavelyev