alexforencich/verilog-ethernet #99: Question about ASYNC_REG

Yuan-Mao, 2021-10-16:
Hi,

I am learning about how to set Vivado properties properly for CDC. When I looked down into the module 'axis_async_fifo', I didn't see any (* ASYNC_REG = "TRUE" *) attached to sync registers, and the tcl files under syn/vivado/ seemed not to set them either. Should I add the ASYNC_REGs myself or they are actually not necessary?

Thanks!

Yuan-Mao, 2021-10-16:
Is there any reason why SHREG_EXTRACTs are not set in tcl files just like how ASYNC_REG does?

Also, is SHREG_EXTRACT = "NO" needed for every synchronizer for Vivado? Although I might miss something again, but the rx/tx_rst_reg don't seem to be set either in RTL or in tcl files? Thank you!

alexforencich, 2021-10-16:
True, I am considering doing something like that, but mainly have to sort out the logistics of where the files would be stored and the like.