alexforencich/verilog-ethernet #108: Unable to Implement Ethernet UDP Protocol on SP605 Evaluation Kit (Xilinx Spartan 6 XC6SLX45t-3FGG484).

wSafwan, 2022-02-25:
Hi,
I am trying to implement Example Design Code for Digilent Atlys (Xilinx Spartan 6 XC6SLX45) on SP605 Evaluation Kit (Xilinx Spartan 6 XC6SLX45t-3FGG484).  
Atlys (Xilinx Spartan 6 XC6SLX45) has a Active Low Reset Push Button, 5 Active Low Push Buttons, 8 Toggle Switches & 8 LEDs, while SP605 (Xilinx Spartan 6 XC6SLX45t-3FGG484) has a Active High Reset Push Button, 4 Active High Push Buttons, 4 Toggle Switches & 4 LEDs. As SP605 has Active High Reset Push Button, I simply invert the Logic for Reset Push Button in fpga.V as shown in the attachment. To deal with the limitations of buttons and switches in SP605, I programmed my SP605 (Xilinx Spartan 6 XC6SLX45t-3FG484) with the following changes one by one.
1) Didn't define Toggle Switches and Push Buttons in UCF.
2) The switches and buttons have been kept in LOW logic by programming (in fpga.V).
3) The switches and buttons have been kept in HIGH logic by programming (in fpga.V).

After getting done with the above changes one by one,  I tried ping command but it didn't work. I also send the UDP data from software "Packet Sender" but again failed. 
After doing the above changes one by one, I tried the ping command, but it didn't work. I'm also sending UDP data from the "Packet Sender" software, but I failed again.

I would be grateful if you could help me with some solution for SP605 (Xilinx Spartan 6 XC6SLX45t-3FGG484).  

I am attaching the following files for your review.
1) Two UCF Files (sp605 UCF, clock)
2) SP605 Hardware User Guide(pdf)
3) My Full Verilog Code (Verilog Ethernet for SP605)
4) Reset Logic (jpg)

[clock UCF.log](https://github.com/alexforencich/verilog-ethernet/files/8141271/clock.UCF.log)
[sp605 UCF.log](https://github.com/alexforencich/verilog-ethernet/files/8141273/sp605.UCF.log)
[SP605 Hardware User Guide.pdf](https://github.com/alexforencich/verilog-ethernet/files/8141275/SP605.Hardware.User.Guide.pdf)
[Verilog Ethernet for SP605.zip](https://github.com/alexforencich/verilog-ethernet/files/8141225/Verilog.Ethernet.for.SP605.zip)
[image]

MJoergen, 2022-02-27:
@wSafwan In line 8 of the sp605 UCF file, you've set the TIMESPEC for the sys_clk_pin to be 100000kHz, but I believe the clock input is 200 MHz. In that case the value should be 200000kHz.

Similarly, in the attached JPG I see in line 98 a value of 10 for CLKIN_PERIOD. If this refers to the sys_clk_pin then the value should be 5 instead of 10.

wSafwan, 2022-02-28:
@MJoergen 
I have done the changes as you highlighted, but still the problem exists.

MJoergen, 2022-03-04:
Regarding the comment on lines 74-75 I have no idea about. That must be for @alexforencich to answer.

However, your fpga.v file looks wrong. As far as I understand, the input clock to the FPGA is changed from 100 MHz to 200 MHz. However, the clock output must be the same in both cases. Therefore, you should read https://www.xilinx.com/support/documentation/user_guides/ug382.pdf particularly around page 71.

The value for CLKDV_DIVIDE should probably be 4, and the value for CLKFX_DIVIDE should probably be unchanged, i.e. 2.0.

You should check in the timing report, that the DCM generates the desired frequencies.

alexforencich, 2022-03-05:
Hmm.  Agreed that something is funny about the DCM configuration, but I'm not sure offhand what the correct configuration is.  It has been a very long time since I have done anything with spartan 6.  As far as I can tell, CLKDV_DIVIDE only affects CLKDV, which is not used here.  Only the CLKFX output is used, so only the CLKFX_MULTIPLY and CLKFX_DIVIDE values are relevant.  But according to the datasheet, both of these must be integers.  You could possibly try 5 and 8, but I cannot tell if this is within the operating range of the DCM.  Another possibility could be to use 5 and 4 just like the ATLYS design, but set CLKIN_DIVIDE_BY_2 to "TRUE".

alexforencich, 2022-03-30:
Well now, accessing a CF card is a whole different ball game.  For that, I would probably recommend looking to some sort of a soft core to handle the CF card, since presumably you'll want a filesystem of some sort on there, and it's much easier to handle that sort of thing from a soft core.  I have never poked at a CF card before from an FPGA, so I'm not going to be able to help you much with that.  But, depending on the speed of the soft core and the CF card, you may want to buffer packets in DRAM before transferring them to the card, so it might make sense to bring up one of the DDR memory interfaces and use a DMA engine to transfer the packet data into DRAM, and potentially also to read the data out of DRAM to send it to the CF card.  All of this is going to be significantly more complex than the UDP/IP stack in this repo.

wSafwan, 2022-03-31:
@alexforencich is it possible and easier to send data from FPGA to PC via UART and log them on PC?

wSafwan, 2022-04-01:
@alexforencich ?

wSafwan, 2022-04-03:
Hi @alexforencich 
Could I use "verilog-uart" ip core in integration with "verilog-ethernet" ip core (developed by you) to log data on PC by sending UDP data from the "Packet Sender" software to FPGA?