alexforencich/verilog-ethernet #109: Arty RTL files

kijukim1, 2022-03-20:
Hello, thank you very much sharing your valuable materials. I have some experience using Vivado HLS but I am still a slow learner for Hardware/FPGA design.

I am trying to run these Verilog files with Vivado.
[image]
However, when I try to run synthesis I always see the message.
[image]

I am just wondering if there is anything that I can do for resolving the problem.

I would be greatly appreciated if you could give me any guidance and tips.
Thank you very much

Best Regards,

alexforencich, 2022-03-20:
See the makefile for the complete file list:

https://github.com/alexforencich/verilog-ethernet/blob/master/example/Arty/fpga/fpga/Makefile

kijukim1, 2022-03-24:
Hello, I really appreciate your help. I saw Run Synthesis and Implementation was successful after I checked the makefile you indicated above.

Unfortunately, I don't have Arty A7 board now, but I got PYNQ Z2 board which is compatible with Arty Z7.
Would you mind telling me any suggestions or ideas to make this code to work on PYNQ Z2/Arty Z7 board?
I can think about getting another Arty A7 board; however, I'd like to run this code with PYNQ Z2/Arty Z7 if possible.

I searched this on the internet, and it says that I need to look at the schematic of the board or get the XDC files.
https://github.com/Digilent/digilent-xdc

Thank you very much again

Best Regards,

alexforencich, 2022-03-27:
The Ethernet port on the Pynq Z2 is connected to the Zynq ARM PS MIO, so it cannot be used from the FPGA PL side as the FPGA cannot directly access the MIO pins.  Most Zynq boards are set up in this way, hence why the ZCU106 only supports operation on the SFP+ ports.  So, if you want to run the verilog-ethernet design on there, then presumably you would have to connect a second Ethernet PHY to the PMOD connectors and use that instead, updating the pin constraints and such accordingly.

kijukim1, 2022-04-28:
Hello, thank you for your sharing your ideas and insights last time.
I got a new Arty A7 board lately. And I tried to run with Arty board with Vivado. And I entered "netcat -u 192.168.1.128 1234" on Vivado.
[image]

But I am seeing that it is not working with the Hardware Manager of Vivado.
Do I have to use a different application such as Command Prompt or another console to run it on the board?

I am so sorry to have bothered you again but I would be really appreciated if you could tell me about any ideas.

Best Regards,

kijukim1, 2022-05-01:
Hello, I'd like to add one more update.
I installed netcat on my computer and typed "netcat -u 192.168.1.128 1234" on cmd.
And I typed Hello; however, I wasn't able to see any text echoed back.
[image]

I just wonder if I did something wrong. 

Thank you very much again.

Best Regards

MehmetSinanTAHTACI, 2022-05-15:
> Hello, I'd like to add one more update. I installed netcat on my computer and typed "netcat -u 192.168.1.128 1234" on cmd. And I typed Hello; however, I wasn't able to see any text echoed back. [image]
> 
> I just wonder if I did something wrong.
> 
> Thank you very much again.
> 
> Best Regards

MehmetSinanTAHTACI, 2022-05-15:
>same thing happen to me. Does not echo at all.

alexforencich, 2022-05-15:
This is probably a network configuration issue.  You have to set up your network config so that the board is accessible at that address, and that may mean disconnecting from a wifi network that shares the 192.168.1.0/24 address space.  Try running wireshark on the interface that the board is connected to and make sure you can see the ARP request going out to the board.

On Linux, I highly recommend using network namespaces to avoid this problem; there is a utility `dev-netns-shell.sh` in the scripts directory that can be used to easily manage a network namespace.  To use the script, run it with the network device as the only argument, then you'll get a shell that can only see that network device.  Add an IP in that rate (i.e. `ip addr add 192.168.1.199/24 dev $dev`), then run netcat.

MehmetSinanTAHTACI, 2022-05-15:
Most probably you are right since the digital logic somehow gives a repeatable result. I will have a look with wireshark.