alexforencich/verilog-ethernet #112: How to implement 1G on Zedboard (Zynq Evaluation & Development Kit)?

wSafwan, 2022-04-17:
Hi @alexforencich,
First of all, thank you for sharing this wonderful library. I have successfully implemented this library code on SP605 (as I discussed with you earlier). Now, I want to implement this on Zedboard which has Marvell 88E1518 PHY. I found that the given library has example designs for Xilinx ZCU102 (Xilinx Zynq UltraScale+ XCZU9EG) & Xilinx ZCU106 (Xilinx Zynq UltraScale+ XCZU7EV). Could you help me implement 1G on Zedboard?

I’m eager to receive your kind feedback.

alexforencich, 2022-04-17:
Unfortunately, the PHY on that board is wired to MIO and hence is only usable from the ARM cores.  See https://github.com/alexforencich/verilog-ethernet/issues/104.  For the ZCU102 and ZCU106, only the 10G interfaces are supported, which are connected to GTH transceivers on the FPGA side.  The gigabit ports on those boards are not supported as they are connected to MIO.

wSafwan, 2022-04-17:
Thanks @alexforencich for your reply. I have some questions. 
1) If PHY is wired to MIO, and after capturing the incoming udp packet through the Processor side, will the captured packet be further processed only on the Processor side, or can it also be processed on the FPGA side too?
2) Will the example designs for Zcu102 & Zcu106 work for Zedboard?

alexforencich, 2022-04-18:
1. Sure, but you would have to do that manually, there is no automatic method to forward packets from PS to PL
2. No, the zedboard has to 10G capable serdes, so those designs cannot be directly ported

One option possibly could be to connect an Ethernet PHY to the FMC connector, such as with https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=71&No=1073&PartNo=2 or https://numato.com/docs/4-port-gigabit-ethernet-fmc-module/

wSafwan, 2022-04-18:
Truly Grateful For All Your Support.

cristalmojo, 2022-06-03:
Hi @wSafwan ,

a few days ago I successfully implemented an example project for the ZedBoard in conjunction with the [Avnet Network FMC Modul](https://www.avnet.com/shop/us/products/avnet-engineering-services/aes-fmc-netw1-g-3074457345635205181/)e. It is an expansion board with two separate RGMII PHYs + ethernet ports. However, [there is a little hardware flaw on the board which prevents the ZedBoard from starting up, but it can be fixed easily as can be seen here](https://www.avnet.com/opasdata/d120001/medias/docus/198/Network_FMC_Errata_200213.pdf).

Let me know if you are interested in trying this out.

wSafwan, 2022-06-06:
@cristalmojo sure, You are most welcome.

cristalmojo, 2022-06-07:
Here you go: You can find the ZedBoard example [here in my forked repository](https://github.com/cristalmojo/verilog-ethernet).

I'm not familiar with your FMC module, but I assume it will work once you adjusted the constraint file. [Probably you can use this one which was provided in an official example design.](https://github.com/fpgadeveloper/ethernet-fmc-max-throughput/blob/master/Vivado/src/constraints/zedboard.xdc)

yoursunny, 2022-06-08:
I'm attempting to use this library on [snickerdoodle black + piSmasher baseboard](https://krtkl.com/snickerdoodle/).
It contains:

* Xilinx Zynq-7020, same as the Zedboard
* Marvell 88E1512 PHY
* PHY signals are wired to Zynq PL pins, except that the PHY RESET signal is wired to Zynq PS MIO pin

This board lacks external clock signal, so that I have to integrate `fpga.v` into a block design, and use Zynq PS to generate the 100MHz clock.
The PHY RESET isn't connected, but I typed `busybox devmem` commands to trigger the reset via MIO pins.

[image]

However, this design doesn't seem to be working.
While the connected PC (192.168.1.1/24) can negotiate 1Gbps/Full, there's no response to ARP.
I also observed the `led[7:0]` signals via AXI GPIO and they are all zeros.
I wonder if someone can give some hints?

yoursunny, 2022-06-08:
@cristalmojo please edit this comment to avoid confusion:
https://github.com/cristalmojo/verilog-ethernet/blob/0886ec3e5f5e8d5bf5c544259e2b6b5ba6c40adf/example/ZedBoard/fpga/rtl/fpga.v#L37

Now I've changed the `reset` polarity into `fpga.v`, but it doesn't make a difference.
[image]

cristalmojo, 2022-06-13:
> @cristalmojo please edit this comment to avoid confusion: https://github.com/cristalmojo/verilog-ethernet/blob/0886ec3e5f5e8d5bf5c544259e2b6b5ba6c40adf/example/ZedBoard/fpga/rtl/fpga.v#L37

Sorry, my bad! Didn't notice this comment when copy-pasting and editing.

Thanks a lot for the hint!