alexforencich/verilog-ethernet #114: ICMP echo

proconsule, 2022-05-19:
Hi, thanks for this great project.
I tried the UDP/IP stack with an ARRIA 10 SX (TERASIC HAN PILOT board) using quartus 21.1.
All works good (UDP echo on port 1234 works very well and arp also)
i know that icmp isn't supported, so i am trying to make it work (have ping makes connectivity test very easy)
i used udp_complete_64 as starting point (since i am using a 10g fiber) but i cant figure out the packet flow.

`
/*
 * Input classifier (ip_protocol)
 */
wire s_select_udp = (ip_rx_ip_protocol == 8'h11);
wire s_select_icmp = (ip_rx_ip_protocol == 8'h01 && ip_rx_ip_eth_type == 16'h0800);
wire s_select_ip = !(s_select_udp || s_select_icmp);
`
i tried with identify that a packet is an icmp ping with this to send to a custom icmp module (using udp_64 as reference)

have you any suggestion of a way to implement it in the right way (inside the ip_complete_64 module maybe?) or is the way i used the right way?

thanks

pradeep-prade, 2022-07-04:
You can implement it on the IP layer. I have done the same and is working fine and tested on the board

proconsule, 2022-07-19:
> You can implement it on the IP layer. I have done the same and is working fine and tested on the board

Can you share the code?

EdwinEstep, 2024-05-06:
> You can implement it on the IP layer. I have done the same and is working fine and tested on the board

I'm also curious. Could you add it to your public fork of `verilog-ethernet`?

JakaBac, 2026-09-07:
@proconsule How did you use the Arria 10 transcievers in your design? Did you use the hard 10Gb Ethernet Enhanced PCS configuration with just the mac from verilog-ethernet or did you use mac+phy from verilog-ethernet?