alexforencich/verilog-ethernet #119: Arty example Testbench simulation shows that the Arty FPGA sends the Host ARP request, but the implemented design on Arty board does not send the ARP request to the Host

zaidtahirbutt, 2022-06-23:
Hi I was comparing the Arty testbench with the implementation of the design using the netcat command and wireshark.
 
Running make verilog-ethernet/example/Arty/fpga/tb/fpga_core/test_fpga_core.py file shows that after receiving UDP data from host, the Arty fpga sends the host an ARP request and after receiving the ARP response, the Arty then sends back the UDP data that it initially received from the host. 

But when I did the experiment with the implemented design on the Arty fpga using the netcat command and wireshark, I saw that the host was sending the Arty fpga an ARP request and got a ARP response back from the Arty FPGA and then sent its UDP data to the Arty fpga and got it back from the fpga.

Why isn't the Arty fpga sending an ARP req to the host in the implemented design as in the testbench? Or am I reading the testbench wrong? 

The output from WireShark is below (Arty fpga has IP 192.168.1.128):

		1	0.000000	PcsCompu_58:9d:6a	Broadcast	ARP	60	Who has 192.168.1.128? Tell 192.168.1.200
		2	0.000244	02:00:00:00:00:00	PcsCompu_58:9d:6a	ARP	60	192.168.1.128 is at 02:00:00:00:00:00
		3	0.000448	192.168.1.200	192.168.1.128	UDP	60	52538 → 1234 Len=3
4	0.000505	192.168.1.128	192.168.1.200	UDP	60	1234 → 52538 Len=3

zaidtahirbutt, 2022-06-23:
I did an experiment with the host having the mac address of the Arty already saved in its arp cache, to see if the Arty fpga sends the host an ARP request as soon as I uploaded the bit file in the arty. The fpga did not send an ARP req to the host.

alexforencich, 2022-06-25:
The host's information is cached on the FPGA when handling the initial ARP request from the host.  So, I suppose this is a minor difference between the testbench and the actual netcat test, as the testbench assumes the MAC is known, and the FPGA has to send an ARP request before it can send the packet.  Possibly the design should be updated to also cache the MAC and IP of incoming packets, but I have not done this because I don't want the ARP cache to fill up with irrelevant information.

zaidtahirbutt, 2022-06-26:
Thanks for the detailed explanation @alexforencich !! It's clear now.

closed 2022-06-26 by zaidtahirbutt