alexforencich/verilog-ethernet #120: How to run tests in tb folder that don't have make files (aren't inside a separate folder) like test_ip_complete.py

zaidtahirbutt, 2022-06-28:
I am able to run the simulation tests inside the tb folder using their make files using command "make WAVES=1". But this is possible only for the tests that are inside their own folders with make files in them.

How to run the tests that don't have make files, like test_ip_complete.py?

I tried running that using command "**pytest test_ip_complete.py**", but first the script wasn't able to import eth_ep.py from the same folder (solved that using sys.path.insert() function). 

But now I am getting this error "**myhdl: Unable to find a `myhdl.vpi' module on the search path.** "

I haven't used myhdl before so it would be great if you could give some directions.

[image]

alexforencich, 2022-06-28:
Those are the old MyHDL tests, and they should probably just be removed as I'm not currently maintaining them.

alexforencich, 2022-06-28:
Anyway, you should be able to simply run those scripts (./script.py or python script.py).  For myhdl.vpi, that needs to be compiled and installed into the proper directory so icarus verilog can find it.  There will be a folder somewhere in /usr/lib called ivl that contains some other vpi files, put myhdl.vpi in there.

closed 2022-06-28 by zaidtahirbutt