alexforencich/verilog-ethernet #123: RGMII clk freq doubt

grey0skull, 2022-07-28:
Hey @alexforencich,

tks for the amazing repo, excuse my ignorance but if RGMII/1G requires a 125MHz clk tx/rx, why the tx data path uses 90MHz clk along with 125MHz@90degress offset? As it's connected to a DDR cell in the end, doesn't it double the tx clk freq to 180MHz? 

Also, it seems a bit tough to match the timing here between the two clocks. I have defined clock groups for most of logic that crosses the asynchronous fifo but for this path I think I can't ignore...is there any suggested trick?
[image]

alexforencich, 2022-09-28:
I have no idea why that clock is labeled clk_90MHz.  Maybe you named it that way in your top-level?  Maybe there is a mistake in how the PLL configuration, or the clock connections?

I did have someone suggest offsetting the phase the other way (270 degrees instead of 90), which may help with timing.  But there would also need to be some adjustments made to the ODDR flip flop, I think the inputs would need to be swapped.  However, I have not had a chance to do any testing with this configuration to make sure it works correctly.

aignacio, 2022-10-03:
Sorry but the clk_90MHz label is coming from your design here:
https://github.com/alexforencich/verilog-ethernet/blob/master/example/NexysVideo/fpga/rtl/fpga.v#L145

alexforencich, 2022-10-03:
Nothing in that file is "clk_90MHz".