alexforencich/verilog-ethernet #124: Illegal argument of type real in math function 'clog2()', expected integral type (Vivado 2022.1)

tux3, 2022-07-29:
Hi,

Running synthesis with Vivado 2022.1 I get the following error log:

```
WARNING: [Synth 8-9887] parameter declaration becomes local in 'ip_mux' with formal parameter declaration list [/source/path/rtl/ip_mux.v:118]
ERROR: [Synth 8-9566] illegal argument of type real in math function 'clog2()', expected integral type [/source/path/rtl/eth_phy_10g_rx_ber_mon.v:62]
WARNING: [Synth 8-9887] parameter declaration becomes local in 'eth_phy_10g_rx_ber_mon' with formal parameter declaration list [/source/path/rtl/eth_phy_10g_rx_ber_mon.v:62]
Failed to read verilog '/source/path/rtl/eth_phy_10g_rx_ber_mon.v'
```

It doesn't seem to like this in `eth_phy_10g_rx_ber_mon.v`:

```verilog
// ...
module eth_phy_10g_rx_ber_mon #
(
    parameter HDR_WIDTH = 2,
    parameter COUNT_125US = 125000/6.4
)
(
// ...
parameter COUNT_WIDTH = $clog2(COUNT_125US);
// ...
```

david-sawatzke, 2022-08-02:
As a workaround I've added `$rtoi(COUNT125US)` and it builds

likewise, 2025-04-23:
In Vivado (at least 2023.2), $rtoi(x) fails if x is not a real, so it cannot be blindly added to all occurrences.