alexforencich/verilog-ethernet #125: Undefined system task '$from_myhdl' during Simulation

dipal004, 2022-08-18:
Hi,
I am trying to simulate the eth_mac_10g_fifo_32 module using the provided testbench. But getting this error . I have tried to find out what the problem is but could not get anything.
Here is the information from the log file.

_Vivado Simulator 2020.2
Copyright 1986-1999, 2001-2020 Xilinx, Inc. All Rights Reserved.
Running: C:/Xilinx/Vivado/2020.2/bin/unwrapped/win64.o/xelab.exe -wto 41c1940df3ce44748ccf6d97c4ccf81a --incr --debug typical --relax --mt 2 -L xil_defaultlib -L unisims_ver -L unimacro_ver -L secureip -L xpm --snapshot test_eth_mac_10g_fifo_32_behav xil_defaultlib.test_eth_mac_10g_fifo_32 xil_defaultlib.glbl -log elaborate.log 
Using 2 slave threads.
Starting static elaboration
Pass Through NonSizing Optimizer
WARNING: [VRFC 10-3091] actual bit length 32 differs from formal bit length 8 for port 's_axis_tid' [C:/Users/sources_1/imports/rtl/eth_mac_10g_fifo.v:415]
WARNING: [VRFC 10-3091] actual bit length 32 differs from formal bit length 8 for port 's_axis_tdest' [C:/Users/sources_1/imports/rtl/eth_mac_10g_fifo.v:416]
WARNING: [VRFC 10-3091] actual bit length 32 differs from formal bit length 8 for port 's_axis_tid' [C:/Users/sources_1/imports/rtl/eth_mac_10g_fifo.v:467]
WARNING: [VRFC 10-3091] actual bit length 32 differs from formal bit length 8 for port 's_axis_tdest' [C:/Users/sources_1/imports/rtl/eth_mac_10g_fifo.v:468]
WARNING: [VRFC 10-5021] port 'ptp_sample_clk' is not connected on this instance [C:/Users/sources_1/imports/tb/test_eth_mac_10g_fifo_32.v:141]
Completed static elaboration
**ERROR: [XSIM 43-4287] "C:/Users/sources_1/imports/tb/test_eth_mac_10g_fifo_32.v" Line 96. Undefined system task '$from_myhdl'_**

If anyone can give any insight on this thing, that would be great.
Thank you

alexforencich, 2022-08-18:
That's an old MyHDL testbench that must be run under Icarus Verilog + MyHDL.  It does not work with the Vivado simulator.  The old testbenches also aren't maintained anymore, so you should use the newer cocotb testbenches.  These should work with more simulators, but unfortunately not the Vivado simulator as the Vivado simulator does not properly implement the Verilog PLI (VPI) and hence cannot be used with cocotb.

dipal004, 2022-08-18:
Hi Alex,
Many thanks for your response. In this case, is there any other way how I can simulate the module in Vivado or there is no option?
The thing is I want to use one of your ethernet module for my own project where I am going to integarte the module with the AXI system bus. I just need to find a way to test the ethernet protocol for the integrated module. As I am new, I am finding it hard to get a good way. 
If you can give me any advice on this that would be great.
Thank you

alexforencich, 2022-08-18:
Unfortunately, unless Xilinx fixes the Vivado simulator PLI, there is no way to run the testbenches in Vivado.  The HDL should simulate just fine in whatever simulator you want to use, but you'll have to write your own testbench for Vivado.

closed 2022-11-02 by alexforencich