alexforencich/verilog-ethernet #129: Zynq 7 timing issue and logic clk question

lk-davidegironi, 2022-09-05:
Hello,

-- question 1
I'm using this core with no problem at all on mine MYIR MYD-Y7Z010 development board (Zynq-7010 based).
Gigabit Ethernet it's provided by KSZ9031 connected as RGMII.
Even if my project works (I've not tested it at Gigabit) I've a timing issue on implementation, on the oddr TX line.
Find it below:

I'm using code from the NexysVideo example to setup clock and board. So my eth_mac_1g_rgmii_fifo it's configured like below:
```
    .IODDR_STYLE("IODDR"),
    .CLOCK_INPUT_STYLE("BUFR"),
    .USE_CLK90("TRUE"),
```
Strange things is that If i leave USE_CLK90 to FALSE, I've no timing issue. But I've read many times Alex reply here on this page telling that for RGMII clk90 should be true in order for the core to work.
What can i check?

-- question 2
I'm trying to use a different clock for logic and rgmii module.
I would like to have main clock at 100Mhz, and RGMII clock of course at 125Mhz.
For this reason I've build a fpga_core with 2 clock
clk125 with reset rst125, running @ 125Mhz
and
clk with reset rst, running @ 100Mhz

The only point I'm using the clk125 it's the eth_mac_1g_rgmii_fifo module, like below:
```
    .gtx_clk(clk125),
    .gtx_clk90(clk125_90),
    .gtx_rst(rst125),
    .logic_clk(clk),
    .logic_rst(rst),
```

Please find fpga.v and fgpa_core.v included for reference.
As you can see I've try to add a new async fifo outside the eth_mac_1g_rgmii_fifo module, but this does not help.

I'm sure there's a valid reason for that but I can not find it.

Thanks!

[timing_summary.txt](https://github.com/alexforencich/verilog-ethernet/files/9490048/timing_summary.txt)
[fpga_test.zip](https://github.com/alexforencich/verilog-ethernet/files/9490055/fpga_test.zip)

alexforencich, 2022-09-28:
If it's a zynq board, is the PHY wired to PS or PL?  If the PHY is wired to PS, then it cannot be used from PL.

lk-davidegironi, 2022-09-29:
Hello Alex, it's wired to PL. I'm able to use that board even with the timing issue, but I have this warning.

lk-davidegironi, 2023-04-11:
Solved! Forget to post there the solution, I've just find out now my open issue here, I'm going to close right now.
For those who are interested, I simply forget to import and instruct the synth engine to run the tcl scripts provided in this repo.
For my use case i need: rgmii_phy_if.tcl, eth_mac_fifo.tcl, eth_mac_1g_rgmii.tcl, axis_async_fifo.tcl, sync_reset.tcl
Simply importing in the constraints section of vivado and setting compile order as above does the job.

closed 2023-04-11 by lk-davidegironi