alexforencich/verilog-ethernet #130: Can run on VCU129

cheungivan, 2022-09-09:
Hi, 
I try to modify it an run it on VCU129 by modify VCU118 file. But i get failure in implementation phase.. Could you help to give us any  idea of such problem?

[DRC PDRC-198] Serial_mode_bitslice_0_connected: SERIAL Mode : The SERIAL mode nibble at BITSLICE_CONTROL cell eth_pcspma/inst/pcs_pma_block_i/gen_io_logic/BaseX_Byte_I_Rx_Nibble/Gen_1.Nibble_I_BitsliceCntrl must have an RX_BITSLICE or RXTX_BITSLICE connected to bitslice position 0 with attribute DATA_TYPE set to 'SERIAL', see Select IO Userguide UG571 for more information.
[runme.log](https://github.com/alexforencich/verilog-ethernet/files/9533976/runme.log)

alexforencich, 2022-09-28:
This is a problem in the Xilinx PCS/PMA core.  There are some settings that need to be adjusted in the core based on the exact placement (which, TBH, is highly annoying to have to do).  You can change this in Vivado by opening up the pcs/pma IP core, changing the parameters, and then looking at the TCL console to see what parameters were changed so you can update the IP TCL script for the PCS/PMA core.  It's possible that some adjustments will also be necessary to the PCS/PMA module instance in the HDL.

akshatramachandran, 2022-10-02:
Hi even I'm looking at implementing the VCU118 design for a VCU128 board but running into similar errors. Can you please guide on how to debug these errors and the approach you used for the VCU118

alexforencich, 2022-10-02:
Like I said, the problem is the SGMII PCS/PMA core on US+ devices when using bitslice LVDS.  Based on the pins being used, several settings need to be adjusted in gig_ethernet_pcs_pma_0.tcl related to the bitslice configuration.  The configuration for the VCU118 is:

    CONFIG.TxLane0_Placement {DIFF_PAIR_2} \
    CONFIG.RxLane0_Placement {DIFF_PAIR_0} \
    CONFIG.Tx_In_Upper_Nibble {0} \

The other boards will require some adjustments, and may need some additional configuration parameters to be set.  See the Xilinx documentation for the SGMII core, bitslice primitives, and the board pin assignments.  And also see https://support.xilinx.com/s/question/0D52E00006iHmN0SAK/1g25g-pcspma-unroutable-internal-net-issues-on-vcu118-xvcu9p?language=en_US.  The main thing you need to look up is the bit slice lane numbers for the TX and RX pin pairs.  Unfortunately, this information is a bit buried.  If you open the elaborated design in Vivado, this will give you a pinout diagram of the chip, and you can look at PKGPIN_NIBBLE_INDEX and also the pin name to figure out upper/lower nibble and diff pair within the nibble.  Once you know the lane numbers, open up the configuration GUI for the core and adjust the settings accordingly.  When you apply the changes, vivado will display the TCL command that contains the new parameters in the TCL console, so update gig_ethernet_pcs_pma_0.tcl based on that.

cheungivan, 2022-10-07:
HI, I already change tcl code as correct pair but problem still exists.. 

set_property -dict [list \
    CONFIG.Standard {SGMII} \
    CONFIG.Physical_Interface {LVDS} \
    CONFIG.Management_Interface {false} \
    CONFIG.SupportLevel {Include_Shared_Logic_in_Core} \
    CONFIG.LvdsRefClk {625} \
    CONFIG.TxLane0_Placement {DIFF_PAIR_1} \
    CONFIG.RxLane0_Placement {DIFF_PAIR_2} \
    CONFIG.Tx_In_Upper_Nibble {0} \
] [get_ips gig_ethernet_pcs_pma_0]

btw, I see you add following setting in other board.. Is it critical to make it work?
set_property LOC BITSLICE_RX_TX_X1Y35 [get_cells -hier -filter {name =~ */lvds_transceiver_mw/serdes_1_to_10_ser8_i/idelay_cal}]

ivancheung2019, 2023-02-02:
Hi, @maheredia 
We can generate bitstream file.. but still communication failure.. 
Is it our PHY chip (TI DP8386) setting incorrect (enable SGMII port)? 
Reg 0x00, 0x0140
Reg 0x10, 0x5848  
Reg 0x1F, 0x4000 
Can you share MIDO setting? We haven't detail datasheet, or sample code...

maheredia, 2023-02-15:
Hi @ivancheung2019 . There is a simple state machine in VCU118 fpga_1g example that performs MDIO writes to configure PHY chip (see **case** statemen in line 389 of **/example/VCU118/fpga_1g/rtl/fpga.v**). 

It seems that VCU128 board has a strap configuration different than the one suggested in DP8386 datasheet, so SGMII mode is not enabled when you power up the board. To solve that, I added an additional register write in that MDIO state machine, just writing 0x0800 to PHYCR (0x0010).