alexforencich/verilog-ethernet #131: How to set DATA_WIDTH=32bit

myqlee, 2022-09-28:
Hi @alexforencich ,
First of all, thank you for sharing this wonderful library. 
I want to implement this project on Alveo U200 card. I follow verilog-ethernet/example/AU200/fpga_10g/README.md,
run make to build, but the DATA_WIDTH of the built project is 64bit, what I want is DATA_WIDTH=32bit. what should I do?

I’m eager to receive your kind feedback.

Thank you,
Eduardo

alexforencich, 2022-09-28:
Unfortunately, this is not currently supported.  The modules in this repo have not been fully parametrized, so for most modules there are separate 8 bit and 64 bit versions, and as a result the full stack currently only supports operation for 8 bits or for 64 bits.  I have rewritten a few modules to support a parametrized data width so the same module can be used from 8 bits to 512 bits, but there is still a lot of work remaining to complete this process, and it is currently a low priority.

myqlee, 2022-09-29:
Thanks for the quick reply.
Here's the thing: I now have an IP core of PCS+PMA, and then implement MAC, that is, synthesis to implement an Ethernet.
I use the eth_mac_10g_fifo module (DATA_WIDTH=64bit) in your project, and write a 64bit to 32bit module by myself (to convert the 64bit axis to the 32bit axis), is it feasible?

myqlee, 2022-09-30:
So I'm going to use the Xilinx PCS/PMA IP core(DATA_WIDTH=32bit), plus your eth_mac_10g_fifo module(DATA_WIDTH=32bit, including eth_mac_10g_fifo.v, eth_mac_10g.v, axis_xgmii_rx_32.v, axis_xgmii_tx_32.v, lfsr.v, axis_async_fifo_adapter.v, axis_async_fifo.v,). 
Thank you again!

myqlee, 2022-09-30:
Well, I have a few more details to ask you:
Q1: Lines 307 and 323 in axis_xgmii_tx_32.v:
xgmii_txd_next = {{3{ETH_PRE}}, XGMII_START};  // line 307
xgmii_txd_next = {ETH_SFD, {3{ETH_PRE}}};         // line 323

7 bytes of synchronous preamble and one byte of SFD, but the codes seem to have only 6 bytes of preamble and one byte of SFD. Why?

Q2: There are i2c_scl/i2c_sda and uart_rxd/uart_txd in fpga.v,
What are the i2c interface and uart interface used for?
If I only use the eth_mac_10g_fifo module, do I need to use the  i2c interface and uart interface?

alexforencich, 2022-09-30:
Q1: the XGMII start control character is sent in place of the first preamble byte.  So it is 7 bytes of preamble.

Q2: they are not used, in many of the example designs there are a few other things connected that are not strictly necessary.  I think those pins on the top-level are simply tied off.

myqlee, 2022-10-08:
Thank you very much!
Another question:
Q3: when I run "python3  test_ axis_ xgmii_ tx_ 64.py", it's OK. But when I run "python3  test_ axis_ xgmii_ tx_ 32.py", it would report an error as follows:
[image]

myqlee, 2022-10-08:
Q4: I  use the two Xilinx PCS/PMA IP core(DATA_WIDTH=32bit): one for RX and another for TX.
So how many eth_mac_10g_fifo(DATA_WIDTH=32bit) modules do I use? One or two?

myqlee, 2022-11-03:
@alexforencich 
I am so sorry to have bothered you again but I would be really appreciated if you could tell me about any ideas.
As I said before, my goal is to implement an Ethernet with xilinx PCS/PMA IP core, plus your eth_mac_10g_fifo module(DATA_WIDTH=32bit, including eth_mac_10g_fifo.v, eth_mac_10g.v, axis_xgmii_rx_32.v, axis_xgmii_tx_32.v, lfsr.v, axis_async_fifo_adapter.v, axis_async_fifo.v,).
Here is the hierarchy tree:
[image]

But, after the arp connection and the tcp connection are successful, the tcp link cannot be maintained for a long time, and repeated replies and retransmissions will occur. See the pcap file from tcpdump. 
[image]

192.168.30.169 is the ip address of U200, while 192.168.30.168 is the ip address of SFC card.

I don't know how to upload the file on this website. I sent the complete .pcap file to alex@alexforencich.com by email.