alexforencich/verilog-ethernet #133: Help to send data using UDP

Juan8UTN, 2022-10-02:
Hi, first of all, thanks for sharing your knowledge and give this IP. I'm currently doing my final project to obtain my Bachelor's Degree in Electronics Engineering and, as a part of the project, i need to send some data using UDP. I was, for several days, reading the differents parts of the code but i wasn't able to send any data. But, the reality is, i really kind of a newbie in the FPGA's world and i got a little lost. Letting aside that much speech, i need to know what modules i should modify to, for the start, send with my computer some string (doesn't matter what) and receive from the FPGA another predefined string as, for example, "Acknowledge". 
Sorry if a i commited some error with english, it is not my first lenguage, and sorry for bother your with this, highly, silly question. 
Greetings from Argentina.

unbtorsten, 2022-10-25:
This is where the the default loopback is implemented: https://github.com/unbtorsten/verilog-ethernet/blob/master/example/Genesys2/fpga_rgmii/rtl/fpga_core.v#L240-L280 (The examples for other boards will be similar, YMMV)

The outbound data is wired up here: https://github.com/unbtorsten/verilog-ethernet/blob/master/example/Genesys2/fpga_rgmii/rtl/fpga_core.v#L270-L274

Juan8UTN, 2022-10-27:
> This is where the the default loopback is implemented: https://github.com/unbtorsten/verilog-ethernet/blob/master/example/Genesys2/fpga_rgmii/rtl/fpga_core.v#L240-L280 (The examples for other boards [...]

Thanks a lot, Torsten. Greetings.

8talha, 2023-01-24:
Hi, Thank you for your guide @unbtorsten . I tried sending data using @unbtorsten  guide. I send my custom data to udp_payload_fifo input and connected it's output to UDP Frame input of udp_complete_inst. The problem I am facing here is "udp_complete_inst" IP is not generating valid true for  Ready for Input signal which is "tx_udp_payload_axis_tready" signal when I insert my data into this IP. but it generates valid true for this signal when I send data from Host to FPGA.
Here is the screenshot of UDP frame Input data signal in idle case(means no transmission of data from host to FPGA)
[image]
Here is the screenshot of UDP frame Input data signal when I send data from Host to fpga
[image]
As you can see 'tx_udp_payload_axis_tready' is switching On when data from Host to fpga comes. Altough it shouldn't behave like this as currently I am sending my custom data from FPGA to Host.
I am also sharing my lines of code where I did modification to send custom data
`
// Loop back UDP
//wire match_cond = rx_udp_dest_port == 1234;
//wire no_match = ~match_cond;

//reg match_cond_reg = 0;
//reg no_match_reg = 0;

//always @(posedge clk) begin
//    if (rst) begin
//        match_cond_reg <= 0;
//        no_match_reg <= 0;
//    end else begin
//        if (rx_udp_payload_axis_tvalid) begin
//            if ((~match_cond_reg & ~no_match_reg) |
//                (rx_udp_payload_axis_tvalid & rx_udp_payload_axis_tready & rx_udp_payload_axis_tlast)) begin
//                match_cond_reg <= match_cond;
//                no_match_reg <= no_match;
//            end
//        end else begin
//            match_cond_reg <= 0;
//            no_match_reg <= 0;
//        end
//    end
//end

//assign tx_udp_hdr_valid = rx_udp_hdr_valid & match_cond;
//assign rx_udp_hdr_ready = (tx_eth_hdr_ready & match_cond) | no_match;
assign tx_udp_hdr_valid = rx_udp_hdr_valid;
assign rx_udp_hdr_ready = tx_eth_hdr_ready;
assign tx_udp_ip_dscp = 0;
assign tx_udp_ip_ecn = 0;
assign tx_udp_ip_ttl = 64;
assign tx_udp_ip_source_ip = local_ip;
assign tx_udp_ip_dest_ip = rx_udp_ip_source_ip;
assign tx_udp_source_port = rx_udp_dest_port;
assign tx_udp_dest_port = rx_udp_source_port;
//assign tx_udp_length = rx_udp_length;
assign tx_udp_length = 16'd50000;

assign tx_udp_checksum = 0;

assign tx_udp_payload_axis_tdata = tx_fifo_udp_payload_axis_tdata;// UDP payload FIFO out data
assign tx_udp_payload_axis_tkeep = tx_fifo_udp_payload_axis_tkeep;
assign tx_udp_payload_axis_tvalid = tx_fifo_udp_payload_axis_tvalid;
//assign tx_fifo_udp_payload_axis_tready = tx_udp_payload_axis_tready;
assign tx_fifo_udp_payload_axis_tready = 1'b1;
assign tx_udp_payload_axis_tlast = tx_fifo_udp_payload_axis_tlast;
assign tx_udp_payload_axis_tuser = tx_fifo_udp_payload_axis_tuser;

//assign rx_fifo_udp_payload_axis_tdata = rx_udp_payload_axis_tdata;//UDP Frame out
//assign rx_fifo_udp_payload_axis_tkeep = rx_udp_payload_axis_tkeep;
//assign rx_fifo_udp_payload_axis_tvalid = rx_udp_payload_axis_tvalid & match_cond_reg;
//assign rx_udp_payload_axis_tready = (rx_fifo_udp_payload_axis_tready & match_cond_reg) | no_match_reg;
//assign rx_fifo_udp_payload_axis_tlast = rx_udp_payload_axis_tlast;
//assign rx_fifo_udp_payload_axis_tuser = rx_udp_payload_axis_tuser;

//Custom data input coming from rx_fifo_udp_payload_axis_tdata1 to udp payload FIFO input
assign rx_fifo_udp_payload_axis_tdata = rx_fifo_udp_payload_axis_tdata1;//custom data input from rx_fifo_udp_payload_axis_tdata1
assign rx_fifo_udp_payload_axis_tkeep = rx_fifo_udp_payload_axis_tkeep1;
assign rx_fifo_udp_payload_axis_tvalid = rx_fifo_udp_payload_axis_tvalid1;
assign rx_udp_payload_axis_tready = rx_fifo_udp_payload_axis_tready;//Tell UDP frame Input FIFO that I am ready
assign rx_fifo_udp_payload_axis_tlast = rx_fifo_udp_payload_axis_tlast1;
assign rx_fifo_udp_payload_axis_tuser = rx_fifo_udp_payload_axis_tuser1;
`