alexforencich/verilog-ethernet #141: Porting to UltraZed-EV?

danisan7, 2023-01-17:
Hello,

First of all thank you for this great repo.

I am trying to port this project to the UltraZed-EV card. For this I plan to use the SFP interface, since the RJ-45 connector goes to the PS part (ARM side). Also, I need the SFP interface to be 1Gigabit.

First I tried it starting from the example for the ZCU106 card, but it is not valid as it is for 10G.

I want to do it for 1G with SFP interface, but I don't see any example that can work for me.

As I have seen in other "issues", I was planning to do it with the Xilinx IP core "1G/2.5G Ethernet PCS/PMA or SGMI (16.2)" from Vivado. But I have several doubts about its configuration.

I am not clear how I have to configure this core IP, for example: in "Physical Interface" select "Device Specific Transceiver" or "LVDS Seral", in "Shared Logic" select "Include Shared Logic in Core" or "Include Shared Logic in Example Design", and I'm also hesitant with the rest of your setup.

Besides configuring the core IP, I'm not sure if I also have to instantiate the coreIP of "UltraScale FPGAs Transceivers Wizard (1.7)" or not.

I'm clueless because of the ZCU106 project I used at the beginning. Since in this project for ZCU106 board the module "eth_xcvr_phy_wrapper" is instantiated for each SFP. This module "eth_xcvr_phy_wrapper" instantiates the "GTH-10BASE-R" coreIP (eth_xcvr_gt_full) and the "eth_phy_10g" module (which would be the "10G/25G Ethernet PCS/PMA PHY"). 

That is, in addition to "1G/2.5G Ethernet PCS/PMA or SGMI (16.2)", would also need "GTH-Gigabit Ethernet" core IP?

Please can you help me?
Thank you!
Dani.

alexforencich, 2023-01-17:
I think a better reference would probably be SGMII on the KC705.  You can also take a look at the NetFPGA SUME design, which is a 10G design but uses the Xilinx 10G PCS/PMA IP core so the setup will be similar.  Anyway, 1G PCS/PMA core is the correct core to use, most likely in 1000BASE-X mode instead of SGMII (unless you're going to use an SGMII SFP which supports 10/100/1000 instead of just 1G).  The options to select depend on how the SFP is wired to the FPGA.  On boards like the VCU108, the PHY chip is connected to normal IO and hence LVDS should be selected.  But an SFP will probably be wired to a GT transceiver site, so "device-specific transceiver" would be the correct option.  Check the board documentation to make this determination.  In general, always select "shared logic in the core" unless you're doing something like packing four channels onto the same quad and hence need to share the PLL - in that case, you'll want to generate two cores, one with the shared logic internal, one with it external, and then make the appropriate cross-connections, see the NetFPGA SUME design for an example of how to do that.  You shouldn't have to generate any GT cores separately when using the PCS/PMA IP core.

alexforencich, 2023-01-17:
Also, maybe I should add a summary of the example designs with some notes on porting to a new board.  Picking the closest FPGA family is usually a good starting point, but in this case there are a few other designs that are probably more useful to look at.

danisan7, 2023-01-18:
Hello!

Thanks for answer so soon.

And thank you very much for all the information!! It has clarified many doubts for me.

With the doubts that you have solved for me, I will try the following. I only intend to use a 1Gigabit SFP interface, which goes only to a Transceiver (I understand that it would be a single channel of the "Quad"). The SFP interface is directly connected to the FPGA, so I'll use "device-specific transceiver" (it's also like that in the PG047 documentation). With "1000BASEX" standard, and the "shared logic in the core" option.

For the rest, I'll use the VCU108 project as a reference.

I'll report back on how it goes.

I don't need any more help at the moment.

Thank you for helping!