alexforencich/verilog-ethernet #143: Modifying code to send custom data from PL to Host

8talha, 2023-02-01:
Hi, Thank you for guide, I tested this code with ZCU111 and now I want to modify this code in "not loopback configuration" for that I follow this issue( #133 ) guide but there are few problems I am facing. I also mentioned my problem in that issue but got no response so creating a new issue here. Here is my problem 
I am sending my custom data **continously** to `udp_payload_fifo` input and connected it's output to UDP Frame input of `udp_complete_inst`. The problem I am facing here is `udp_complete_inst` IP is not generating true for  Ready for Input signal which is `tx_udp_payload_axis_tready` signal when I insert my data into this IP. but it generates Ready for Input for this signal when I send data from Host to FPGA.
Here is the screenshot of UDP frame Input data signal in idle case (means no transmission of data from host to FPGA)
[image]
Here is the screenshot of UDP frame Input data signal when I send data from Host to fpga
[image]
As you can see `tx_udp_payload_axis_tready` is switching On when data from Host to fpga comes. Although it shouldn't behave like this as currently I am sending my custom data from FPGA to Host continously.
I am also sharing my lines of code where I did modification to send custom data
```
`
// Loop back UDP
//wire match_cond = rx_udp_dest_port == 1234;
//wire no_match = ~match_cond;

//reg match_cond_reg = 0;
//reg no_match_reg = 0;

//always @(posedge clk) begin
//    if (rst) begin
//        match_cond_reg <= 0;
//        no_match_reg <= 0;
//    end else begin
//        if (rx_udp_payload_axis_tvalid) begin
//            if ((~match_cond_reg & ~no_match_reg) |
//                (rx_udp_payload_axis_tvalid & rx_udp_payload_axis_tready & rx_udp_payload_axis_tlast)) begin
//                match_cond_reg <= match_cond;
//                no_match_reg <= no_match;
//            end
//        end else begin
//            match_cond_reg <= 0;
//            no_match_reg <= 0;
//        end
//    end
//end

//assign tx_udp_hdr_valid = rx_udp_hdr_valid & match_cond;
//assign rx_udp_hdr_ready = (tx_eth_hdr_ready & match_cond) | no_match;
assign tx_udp_hdr_valid = rx_udp_hdr_valid;
assign rx_udp_hdr_ready = tx_eth_hdr_ready;
assign tx_udp_ip_dscp = 0;
assign tx_udp_ip_ecn = 0;
assign tx_udp_ip_ttl = 64;
assign tx_udp_ip_source_ip = local_ip;
assign tx_udp_ip_dest_ip = rx_udp_ip_source_ip;
assign tx_udp_source_port = rx_udp_dest_port;
assign tx_udp_dest_port = rx_udp_source_port;
//assign tx_udp_length = rx_udp_length;
assign tx_udp_length = 16'd50000;

assign tx_udp_checksum = 0;

assign tx_udp_payload_axis_tdata = tx_fifo_udp_payload_axis_tdata;// UDP payload FIFO out data
assign tx_udp_payload_axis_tkeep = tx_fifo_udp_payload_axis_tkeep;
assign tx_udp_payload_axis_tvalid = tx_fifo_udp_payload_axis_tvalid;
//assign tx_fifo_udp_payload_axis_tready = tx_udp_payload_axis_tready;
assign tx_fifo_udp_payload_axis_tready = 1'b1;
assign tx_udp_payload_axis_tlast = tx_fifo_udp_payload_axis_tlast;
assign tx_udp_payload_axis_tuser = tx_fifo_udp_payload_axis_tuser;

//assign rx_fifo_udp_payload_axis_tdata = rx_udp_payload_axis_tdata;//UDP Frame out
//assign rx_fifo_udp_payload_axis_tkeep = rx_udp_payload_axis_tkeep;
//assign rx_fifo_udp_payload_axis_tvalid = rx_udp_payload_axis_tvalid & match_cond_reg;
//assign rx_udp_payload_axis_tready = (rx_fifo_udp_payload_axis_tready & match_cond_reg) | no_match_reg;
//assign rx_fifo_udp_payload_axis_tlast = rx_udp_payload_axis_tlast;
//assign rx_fifo_udp_payload_axis_tuser = rx_udp_payload_axis_tuser;

//Custom data input coming from rx_fifo_udp_payload_axis_tdata1 to udp payload FIFO input
assign rx_fifo_udp_payload_axis_tdata = rx_fifo_udp_payload_axis_tdata1;//custom data input from rx_fifo_udp_payload_axis_tdata1
assign rx_fifo_udp_payload_axis_tkeep = rx_fifo_udp_payload_axis_tkeep1;
assign rx_fifo_udp_payload_axis_tvalid = rx_fifo_udp_payload_axis_tvalid1;
assign rx_udp_payload_axis_tready = rx_fifo_udp_payload_axis_tready;//Tell UDP frame Input FIFO that I am ready
assign rx_fifo_udp_payload_axis_tlast = rx_fifo_udp_payload_axis_tlast1;
assign rx_fifo_udp_payload_axis_tuser = rx_fifo_udp_payload_axis_tuser1;
`
```

alexforencich, 2023-02-01:
You cannot simply dump a bunch of payload data into the stack, you have to generate both headers and payloads and send them into the stack together.  So effectively what you've done is create a design where the headers are looped back, but the payloads will come from somewhere else.  Which is most likely not what you want.

8talha, 2023-02-02:
Hi, Thank you for pointing out mistake, so I also generated udp header but still I am unable to transmit data from PL to Host, and my loopback configuration is removed now because no data is coming to host even when I send data from host to FPGA.
Here is the the default UDP frame input protocol signal pattern
[image]
Here is my custom UDP frame Input protocol pattern
[image]

In my pattern you can see `tx_udp_payload_axis_tready` is not true, can you tell me what I am doing wrong in this?

unbtorsten, 2023-02-02:
Hi and sorry for not replying earlier. I second Alex' diagnosis of the previous issue with header/payload not being aligned. From what I see in your postings, you have not connected `assign tx_fifo_udp_payload_axis_tready`. Hence, the IP does not communicate to your FIFO that it is ready to accept data.

8talha, 2023-02-03:
Thank you for the reply @unbtorsten. I have connected `tx_fifo_udp_payload_axis_tready` to my custom data generation IP which is currently giving constant HIGH to this signal, as you can see in my below mentioned ILA response of `udp_payload_fifo` in and out.
UDP Payload FIFO Input response 
[image]
UDP Payload FIFO Output response 
[image]
 I think I am missing something for `udp_complete_64` as `tx_udp_payload_axis_tready` is not getting ready as you can see in my above mentioned comment which means this IP is not ready to accept my data. The custom signals which I am feeding to this IP are `tx_udp_hdr_valid` and AXI protocol related signals `tx_udp_payload_axis_*`.

unbtorsten, 2023-02-03:
> Thank you for the reply @unbtorsten. I have connected `tx_fifo_udp_payload_axis_tready` to my custom data generation IP which is currently giving constant HIGH to this signal, [...]

If I understand you correctly, you are using the `_tready` signal to indicate data being ready for sending. However, this status is communicated through `_tvalid`. `_tready` indicates a device's willingness to accept data. In other words, the udp module indicates, that it is ready to accept and process data.

zhangjialiang-tt, 2023-04-16:
I'm doing the same thing.just modify `s_udp_` port will be worked.

like:
localparam SYS_CLOCK = 125000000;
always @ ( posedge clk ) begin
    if(rst) begin
        test_data <= 'd0;
    end
    else if( test_data ==SYS_CLOCK-'d1 )begin
        test_data <= 'd0;
    end
    else begin
        test_data <= test_data + 'd1;
    end
end

assign tx_udp_hdr_valid = test_data=='d1;
assign tx_udp_ip_dscp = 0;
assign tx_udp_ip_ecn = 0;
assign tx_udp_ip_ttl = 64;
assign tx_udp_ip_source_ip = local_ip;//c0a80180
assign tx_udp_ip_dest_ip = dest_ip;//c0a80103
assign tx_udp_source_port = 16'd1234;//rx_udp_dest_port;
assign tx_udp_dest_port = 16'd1234;//rx_udp_source_port;
assign tx_udp_length = 100+8;
assign tx_udp_payload_axis_tdata = test_data;
assign tx_udp_payload_axis_tvalid = test_data>='d100&&test_data<'d200;
assign tx_fifo_udp_payload_axis_tready = tx_udp_payload_axis_tready;
assign tx_udp_payload_axis_tlast = test_data=='d200-'d1;
assign tx_udp_payload_axis_tuser = 1'b0;

ahmetserdar, 2023-06-15:
@8talha did you solve problem? When i try to send UDP data from FPGA to HOST , i have this issue."tx_tready" signal gives to zero that indicates "fpga" IP is not ready to accept data.  What should i do?