alexforencich/verilog-ethernet #146: Porting verilog-ethernet to KR260

vmayoral, 2023-02-13:
*(This first comment will be updated as the porting efforts evolve and will summarize results so that others can benefit from it.)*

Hi @alexforencich and all,

[image]

First off, congrats for an impresive piece. The project is not only well maintained but has grown a significant community around it. Very impressive. Congrats and kudos to @alexforencich. I'm currently involved into a project considering to port Corundum into the AMD [KR260](https://www.xilinx.com/products/som/kria/kr260-robotics-starter-kit.html). This board leverages an AMD Zynq UltraScale+ SoC and exposes various Ethernet (Gb RGMII PL and PS) interfaces and an SPF+ connector which makes it a pretty target for benchmarking.

For now, our endeavour aims to bring ethernet-verilog up to speed into the [KR260](https://www.xilinx.com/products/som/kria/kr260-robotics-starter-kit.html) and produce some latency performance benchmarking comparisons by leveraging a low overhead and real-time capable tracer (we'll be using [LTTng](https://lttng.org/) on the PS side, and Vitis Profiler on the PL side). 

By doing so, we hope to contribute to this project by `1)` bringing a usable port of ethernet-verilog (Corundum) for a popular FPGA SoC SOM, `2)` provide a fair and reproducible latency benchmark that compares UDP/IP networking performance (L1-L2-L3-L4) across various setups.

After reading through [^1], it appears that creating example designs for [verilog-ethernet](https://github.com/alexforencich/verilog-ethernet) is the first recommended step. This ticket will thereby track progress on this regard, **porting verilog-ethernet to KR260**. The list below aims to lay out some of the milestones neccessary to achieve the porting goals.

<details><summary>Sources considered</summary>

- Source code: https://github.com/alexforencich/verilog-ethernet
- Paper https://cseweb.ucsd.edu/~snoeren/papers/corundum-fccm20.pdf
- Future of corundum https://github.com/corundum/corundum/wiki/Future-of-Corundum
- Corundum overview https://docs.corundum.io/en/latest/modules/overview.html
- Corundum porting guide https://docs.corundum.io/en/latest/porting.html
- Corundum operations, describes reception and transmission operations https://docs.corundum.io/en/latest/operations.html (:warning: useful yet complex to follow)

</details>

<details><summary>Related issues and past experiences</summary>

- https://github.com/alexforencich/verilog-ethernet/issues/16
- https://github.com/alexforencich/verilog-ethernet/issues/106
- https://github.com/alexforencich/verilog-ethernet/issues/64

</details>

### Environment used
- Vitis/Vivado release: `2022.1`
- branch of [repo](https://github.com/alexforencich/verilog-ethernet) used: `master`

### Milestones:
- [x] verilog-ethernet in ZCU102 ([link](https://github.com/alexforencich/verilog-ethernet/tree/master/example/ZCU102/fpga) to example)
  - [x] inspect source code of `10G BASE-R PHY IP` example
  - [x] Understand build system setup and hooks to Vivado
  - [ ] Build ethernet-verilog for ZCU102 (`10G BASE-R PHY IP core`)
  - [ ] Test `10G BASE-R PHY`
- [ ] verilog-ethernet in KR260
  - [ ] Port `10G BASE-R PHY` example to the SPF+ connector
  - [ ] Bring up a `1G PHY` example to the SPF+ connector
  - [ ] Bring up a `1G MAC` example into one of the PL Ethernet PHYs
- [ ] Latency performance benchmarking
  - [ ] Produce CPU baseline by using Ethernet PHYs connected to the PS and the Linux Networking Stack (LNS)
  - [ ] Benchmark `10G BASE-R PHY` example to the SPF+ connector
  - [ ] Benchmark `1G PHY` example to the SPF+ connector
  - [ ] Benchmark `1G MAC` example into one of the PL Ethernet PHYs

[^1]: Corundum porting guide https://docs.corundum.io/en/latest/porting.html

vmayoral, 2023-02-13:
Bumped into a build error while attempting to create IP for ZCU102:

```bash
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2022.1/data/ip'.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_full' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_full/eth_xcvr_gt_full.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_full' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_full/eth_xcvr_gt_full_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_full' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_full/eth_xcvr_gt_full_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_full' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_full/eth_xcvr_gt_full_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_full' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_full/eth_xcvr_gt_full_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_channel' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_channel/eth_xcvr_gt_channel.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_channel' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_channel/eth_xcvr_gt_channel_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_channel' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_channel/eth_xcvr_gt_channel_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_channel' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_channel/eth_xcvr_gt_channel_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'eth_xcvr_gt_channel' generated file not found '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga/fpga.gen/sources_1/ip/eth_xcvr_gt_channel/eth_xcvr_gt_channel_sim_netlist.vhdl'. Please regenerate to continue.
# reset_run impl_1
# launch_runs -jobs 4 impl_1
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
eth_xcvr_gt_full_synth_1
eth_xcvr_gt_channel_synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Mon Feb 13 17:22:56 2023...
make[1]: *** [../common/vivado.mk:115: fpga.runs/impl_1/fpga_routed.dcp] Error 1
make[1]: Leaving directory '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga'
make: *** [Makefile:14: fpga] Error 2
```

<details><summary>Full prompt of error</summary>

```bash
victor@accelerationrobotics:~/verilog-ethernet/example/ZCU102/fpga$ make
cd fpga && make
make[1]: Entering directory '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga'
rm -rf defines.v
touch defines.v
for x in ; do echo '`define' $x >> defines.v; done
echo "create_project -force -part xczu9eg-ffvb1156-2-e fpga" > create_project.tcl
echo "add_files -fileset sources_1 defines.v  ../rtl/fpga.v  ../rtl/fpga_core.v  ../rtl/eth_xcvr_phy_wrapper.v  ../rtl/debounce_switch.v  ../rtl/sync_signal.v  ../lib/eth/rtl/eth_mac_10g_fifo.v  ../lib/eth/rtl/eth_mac_10g.v  ../lib/eth/rtl/axis_xgmii_rx_64.v  ../lib/eth/rtl/axis_xgmii_tx_64.v  ../lib/eth/rtl/eth_phy_10g.v  ../lib/eth/rtl/eth_phy_10g_rx.v  ../lib/eth/rtl/eth_phy_10g_rx_if.v  ../lib/eth/rtl/eth_phy_10g_rx_frame_sync.v  ../lib/eth/rtl/eth_phy_10g_rx_ber_mon.v  ../lib/eth/rtl/eth_phy_10g_rx_watchdog.v  ../lib/eth/rtl/eth_phy_10g_tx.v  ../lib/eth/rtl/eth_phy_10g_tx_if.v  ../lib/eth/rtl/xgmii_baser_dec_64.v  ../lib/eth/rtl/xgmii_baser_enc_64.v  ../lib/eth/rtl/lfsr.v  ../lib/eth/rtl/eth_axis_rx.v  ../lib/eth/rtl/eth_axis_tx.v  ../lib/eth/rtl/udp_complete_64.v  ../lib/eth/rtl/udp_checksum_gen_64.v  ../lib/eth/rtl/udp_64.v  ../lib/eth/rtl/udp_ip_rx_64.v  ../lib/eth/rtl/udp_ip_tx_64.v  ../lib/eth/rtl/ip_complete_64.v  ../lib/eth/rtl/ip_64.v  ../lib/eth/rtl/ip_eth_rx_64.v  ../lib/eth/rtl/ip_eth_tx_64.v  ../lib/eth/rtl/ip_arb_mux.v  ../lib/eth/rtl/arp.v  ../lib/eth/rtl/arp_cache.v  ../lib/eth/rtl/arp_eth_rx.v  ../lib/eth/rtl/arp_eth_tx.v  ../lib/eth/rtl/eth_arb_mux.v  ../lib/eth/lib/axis/rtl/arbiter.v  ../lib/eth/lib/axis/rtl/priority_encoder.v  ../lib/eth/lib/axis/rtl/axis_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo_adapter.v  ../lib/eth/lib/axis/rtl/sync_reset.v" >> create_project.tcl
echo "set_property top fpga [current_fileset]" >> create_project.tcl
echo "add_files -fileset constrs_1  ../fpga.xdc  ../lib/eth/syn/vivado/eth_mac_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/axis_async_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/sync_reset.tcl" >> create_project.tcl
for x in ; do echo "import_ip $x" >> create_project.tcl; done
for x in  ../ip/eth_xcvr_gt.tcl; do echo "source $x" >> create_project.tcl; done
vivado -nojournal -nolog -mode batch -source create_project.tcl

****** Vivado v2022.1 (64-bit)
[... cut ...]
ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
eth_xcvr_gt_full_synth_1
eth_xcvr_gt_channel_synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.

INFO: [Common 17-206] Exiting Vivado at Mon Feb 13 17:22:56 2023...
make[1]: *** [../common/vivado.mk:115: fpga.runs/impl_1/fpga_routed.dcp] Error 1
make[1]: Leaving directory '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga'
make: *** [Makefile:14: fpga] Error 2
```

</details>

vmayoral, 2023-02-13:
I looked further into the logs as adviced and here's what I found:

```bash
victor@accelerationrobotics:~/verilog-ethernet/example/ZCU102/fpga$ du -a| grep log
4	./fpga/fpga.runs/eth_xcvr_gt_full_synth_1/runme.log
4	./fpga/fpga.runs/eth_xcvr_gt_channel_synth_1/runme.log
victor@accelerationrobotics:~/verilog-ethernet/example/ZCU102/fpga$ cat ./fpga/fpga.runs/eth_xcvr_gt_channel_synth_1/runme.log

*** Running vivado
    with args -log eth_xcvr_gt_channel.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source eth_xcvr_gt_channel.tcl

****** Vivado v2022.1 (64-bit)
  **** SW Build 3526262 on Mon Apr 18 15:47:01 MDT 2022
  **** IP Build 3524634 on Mon Apr 18 20:55:01 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source eth_xcvr_gt_channel.tcl -notrace
Command: synth_design -top eth_xcvr_gt_channel -part xczu9eg-ffvb1156-2-e -incremental_mode off -mode out_of_context
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xczu9eg'
WARNING: [Common 17-348] Failed to get the license for feature 'Synthesis' and/or device 'xczu9eg'. Explanation: The license feature Synthesis could not be found.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ".
0 Infos, 1 Warnings, 0 Critical Warnings and 1 Errors encountered.
synth_design failed
ERROR: [Common 17-345] A valid license was not found for feature 'Synthesis' and/or device 'xczu9eg'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ". If you are using a license server, verify that the license server is up and running a version of the xilinx daemon that is compatible with the version of Xilinx software that you are using. Note: Vivado 2021.1 and later versions require upgrading your license server tools to the Flex 11.17.2.0 versions. Please confirm with your license admin that the correct version of the license server tools are installed.
INFO: [Common 17-206] Exiting Vivado at Mon Feb 13 20:31:35 2023...
```

It seems a license issue. Will re-try things once the license is set up appropriately.

alexforencich, 2023-02-13:
Ah yep, looks like Vivado needs a license for that part.  The Kira SoMs do not require a license, correct?

vmayoral, 2023-02-14:
> Ah yep, looks like Vivado needs a license for that part. The Kira SoMs do not require a license, correct?

It appears to be the case. Same as with the ZCU102, license is required.  For completeness, ZCU102 built successfully after setting up a license:

<details><summary>built prompt</summary>

```bash
victor@accelerationrobotics:~/verilog-ethernet/example/ZCU102/fpga$ make
cd fpga && make
make[1]: Entering directory '/home/victor/verilog-ethernet/example/ZCU102/fpga/fpga'
rm -rf defines.v
touch defines.v
for x in ; do echo '`define' $x >> defines.v; done
echo "create_project -force -part xczu9eg-ffvb1156-2-e fpga" > create_project.tcl
echo "add_files -fileset sources_1 defines.v  ../rtl/fpga.v  ../rtl/fpga_core.v  ../rtl/eth_xcvr_phy_wrapper.v  ../rtl/debounce_switch.v  ../rtl/sync_signal.v  ../lib/eth/rtl/eth_mac_10g_fifo.v  ../lib/eth/rtl/eth_mac_10g.v  ../lib/eth/rtl/axis_xgmii_rx_64.v  ../lib/eth/rtl/axis_xgmii_tx_64.v  ../lib/eth/rtl/eth_phy_10g.v  ../lib/eth/rtl/eth_phy_10g_rx.v  ../lib/eth/rtl/eth_phy_10g_rx_if.v  ../lib/eth/rtl/eth_phy_10g_rx_frame_sync.v  ../lib/eth/rtl/eth_phy_10g_rx_ber_mon.v  ../lib/eth/rtl/eth_phy_10g_rx_watchdog.v  ../lib/eth/rtl/eth_phy_10g_tx.v  ../lib/eth/rtl/eth_phy_10g_tx_if.v  ../lib/eth/rtl/xgmii_baser_dec_64.v  ../lib/eth/rtl/xgmii_baser_enc_64.v  ../lib/eth/rtl/lfsr.v  ../lib/eth/rtl/eth_axis_rx.v  ../lib/eth/rtl/eth_axis_tx.v  ../lib/eth/rtl/udp_complete_64.v  ../lib/eth/rtl/udp_checksum_gen_64.v  ../lib/eth/rtl/udp_64.v  ../lib/eth/rtl/udp_ip_rx_64.v  ../lib/eth/rtl/udp_ip_tx_64.v  ../lib/eth/rtl/ip_complete_64.v  ../lib/eth/rtl/ip_64.v  ../lib/eth/rtl/ip_eth_rx_64.v  ../lib/eth/rtl/ip_eth_tx_64.v  ../lib/eth/rtl/ip_arb_mux.v  ../lib/eth/rtl/arp.v  ../lib/eth/rtl/arp_cache.v  ../lib/eth/rtl/arp_eth_rx.v  ../lib/eth/rtl/arp_eth_tx.v  ../lib/eth/rtl/eth_arb_mux.v  ../lib/eth/lib/axis/rtl/arbiter.v  ../lib/eth/lib/axis/rtl/priority_encoder.v  ../lib/eth/lib/axis/rtl/axis_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo_adapter.v  ../lib/eth/lib/axis/rtl/sync_reset.v" >> create_project.tcl
echo "set_property top fpga [current_fileset]" >> create_project.tcl
echo "add_files -fileset constrs_1  ../fpga.xdc  ../lib/eth/syn/vivado/eth_mac_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/axis_async_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/sync_reset.tcl" >> create_project.tcl
for x in ; do echo "import_ip $x" >> create_project.tcl; done
for x in  ../ip/eth_xcvr_gt.tcl; do echo "source $x" >> create_project.tcl; done
vivado -nojournal -nolog -mode batch -source create_project.tcl

****** Vivado v2022.1 (64-bit)
  **** SW Build 3526262 on Mon Apr 18 15:47:01 MDT 2022
  **** IP Build 3524634 on Mon Apr 18 20:55:01 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source create_project.tcl
# create_project -force -part xczu9eg-ffvb1156-2-e fpga
# add_files -fileset sources_1 defines.v  ../rtl/fpga.v  ../rtl/fpga_core.v  ../rtl/eth_xcvr_phy_wrapper.v  ../rtl/debounce_switch.v  ../rtl/sync_signal.v  ../lib/eth/rtl/eth_mac_10g_fifo.v  ../lib/eth/rtl/eth_mac_10g.v  ../lib/eth/rtl/axis_xgmii_rx_64.v  ../lib/eth/rtl/axis_xgmii_tx_64.v  ../lib/eth/rtl/eth_phy_10g.v  ../lib/eth/rtl/eth_phy_10g_rx.v  ../lib/eth/rtl/eth_phy_10g_rx_if.v  ../lib/eth/rtl/eth_phy_10g_rx_frame_sync.v  ../lib/eth/rtl/eth_phy_10g_rx_ber_mon.v  ../lib/eth/rtl/eth_phy_10g_rx_watchdog.v  ../lib/eth/rtl/eth_phy_10g_tx.v  ../lib/eth/rtl/eth_phy_10g_tx_if.v  ../lib/eth/rtl/xgmii_baser_dec_64.v  ../lib/eth/rtl/xgmii_baser_enc_64.v  ../lib/eth/rtl/lfsr.v  ../lib/eth/rtl/eth_axis_rx.v  ../lib/eth/rtl/eth_axis_tx.v  ../lib/eth/rtl/udp_complete_64.v  ../lib/eth/rtl/udp_checksum_gen_64.v  ../lib/eth/rtl/udp_64.v  ../lib/eth/rtl/udp_ip_rx_64.v  ../lib/eth/rtl/udp_ip_tx_64.v  ../lib/eth/rtl/ip_complete_64.v  ../lib/eth/rtl/ip_64.v  ../lib/eth/rtl/ip_eth_rx_64.v  ../lib/eth/rtl/ip_eth_tx_64.v  ../lib/eth/rtl/ip_arb_mux.v  ../lib/eth/rtl/arp.v  ../lib/eth/rtl/arp_cache.v  ../lib/eth/rtl/arp_eth_rx.v  ../lib/eth/rtl/arp_eth_tx.v  ../lib/eth/rtl/eth_arb_mux.v  ../lib/eth/lib/axis/rtl/arbiter.v  ../lib/eth/lib/axis/rtl/priority_encoder.v  ../lib/eth/lib/axis/rtl/axis_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo.v  ../lib/eth/lib/axis/rtl/axis_async_fifo_adapter.v  ../lib/eth/lib/axis/rtl/sync_reset.v
# set_property top fpga [current_fileset]
# add_files -fileset constrs_1  ../fpga.xdc  ../lib/eth/syn/vivado/eth_mac_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/axis_async_fifo.tcl  ../lib/eth/lib/axis/syn/vivado/sync_reset.tcl
# source ../ip/eth_xcvr_gt.tcl
## set base_name {eth_xcvr_gt}
## set preset {GTH-10GBASE-R}
## set freerun_freq {125}
## set line_rate {10.3125}
## set refclk_freq {156.25}
## set qpll_fracn [expr {int(fmod($line_rate*1000/2 / $refclk_freq, 1)*pow(2, 24))}]
## set user_data_width {64}
## set int_data_width {32}
## set extra_ports [list]
## set extra_pll_ports [list {qpll0lock_out}]
## set config [dict create]
## dict set config TX_LINE_RATE $line_rate
## dict set config TX_REFCLK_FREQUENCY $refclk_freq
## dict set config TX_QPLL_FRACN_NUMERATOR $qpll_fracn
## dict set config TX_USER_DATA_WIDTH $user_data_width
## dict set config TX_INT_DATA_WIDTH $int_data_width
## dict set config RX_LINE_RATE $line_rate
## dict set config RX_REFCLK_FREQUENCY $refclk_freq
## dict set config RX_QPLL_FRACN_NUMERATOR $qpll_fracn
## dict set config RX_USER_DATA_WIDTH $user_data_width
## dict set config RX_INT_DATA_WIDTH $int_data_width
[... cut ...]
INFO: [runtcl-4] Executing : report_timing_summary -max_paths 10 -report_unconstrained -file fpga_timing_summary_routed.rpt -pb fpga_timing_summary_routed.pb -rpx fpga_timing_summary_routed.rpx -warn_on_violation
INFO: [Timing 38-91] UpdateTimingParams: Speed grade: -2, Temperature grade: E, Delay Type: min_max.
INFO: [Timing 38-191] Multithreading enabled for timing update using a maximum of 6 CPUs
WARNING: [Timing 38-436] There are set_bus_skew constraint(s) in this design. Please run report_bus_skew to ensure that bus skew requirements are met.
INFO: [runtcl-4] Executing : report_i

alexforencich, 2023-02-14:
Took a look at the KR260 schematic; it looks like porting the design from the ZCU102 should mainly be a matter of updating the pin constraints and deleting 3 of the SFP+ channels.  Although, it looks like there isn't an external 125 MHz clock (the GTH transceivers need a free-running clock of some sort) so maybe you can borrow the setup I have for the VCU1525 to take the transceiver ref clock and convert that to 125 MHz, see https://github.com/alexforencich/verilog-ethernet/blob/master/example/VCU1525/fpga_10g/rtl/fpga.v#L364 .  I suppose using a PLL to convert to 125 MHz is not strictly necessary, the GTH transceivers would probably work fine with using the 156.25 MHz ref clock directly, but I like putting that PLL in so things are consistent.  You'll need to adjust the PLL settings for 156.25 MHz as the VCU1525 uses 161.1328125 MHz.  Looks like 156.25 * 8/10 = 125, so D = 1 and M = 8 for Fvco 1250 MHz, output divide = 10 to get 125 MHz, and set the input period to 6.4.

Also, it looks like the 156.25 MHz oscillator is fixed, which is a little unfortunate.  Fine for PTP, but something like sync-e or white rabbit would need an adjustable oscillator like an si570 or PLL chip with ppb-resolution adjustment capability.

alexforencich, 2023-02-16:
Actually, on taking another look at the KR260 schematic, it looks like there are 25 MHz clocks provided to the PL on two different pins (see page 16 on the sch).  So, you could feed that into a PLL and covert it to 125 MHz, instead of using the 156.25 MHz from the ref osc.  Either method would work, but if you want to make a 1G design it might make more sense to use one of those 25 MHz clocks instead of the GTH ref clock.

vmayoral, 2023-03-06:
All right, so spent the last few weeks understanding and porting the code for the KR260. The branch that currently reflects my latest status is https://github.com/vmayoral/verilog-ethernet/tree/kr260. I've gone through quite a few iterations while using the following resources:

- Schematics
  - ZCU102 schematics ([0381701r01.pdf](/uploads/5a4075acff2bf803dbfa16fcaed72476/0381701r01.pdf))
  - KR260 schematics ([038-05101-01_sck-kr_reva02_SKIT_20211015.pdf](/uploads/f95059c013722a01987e27c947bbec2c/038-05101-01_sck-kr_reva02_SKIT_20211015.pdf))
- User Guides
  - [Kria SOM Carrier Card Design Guide](https://docs.xilinx.com/r/en-US/ug1091-carrier-card-design/Introduction)
  - [Kria KR260 Starter Kit User Guide](https://docs.xilinx.com/r/en-US/ug1092-kr260-starter-kit/Interfaces)
  - [Kria K26 SOM Datasheet](https://china.xilinx.com/content/dam/xilinx/support/documents/data_sheets/ds987-k26-som.pdf)
  - [ZCU102 Evaluation Board User Guide](https://docs.xilinx.com/v/u/en-US/ug1182-zcu102-eval-bd)
  - [K26 SOM XDC file](https://www.xilinx.com/member/forms/download/design-license.html?cid=29e0261a-9532-4a47-bb06-38c83bbbb8c0&filename=xtp685-kria-k26-som-xdc.zip)

After jumping over quite a few hurdles, I bumped into DRC errors about a week ago and have been fighting with them ever since. Dropping my struggles in here for completeness:

**Struggle 1**
<details><summary>:x: (summary) LVCMOS33 I/O standard is not supported for banks of type High Performance</summary>

```
Attempting to get a license for feature 'Implementation' and/or device 'xczu9eg'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xczu9eg'
INFO: [Common 17-86] Your Implementation license expires in 15 day(s)
INFO: [DRC 23-27] Running DRC with 6 threads
INFO: [Vivado_Tcl 4-198] DRC finished with 0 Errors
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
Running DRC as a precondition to command place_design
INFO: [DRC 23-27] Running DRC with 6 threads
ERROR: [DRC BIVB-1] Bank IO standard Support: Bank 66 has incompatible IO(s) because: The LVCMOS33 I/O standard is not supported for banks of type High Performance.  Move the following ports or change their properties:
sfp0_tx_disable_b
ERROR: [DRC PLHDIO-5] HDIO DRC Checks: The following IO terminals need to be placed in HIGH_DENSITY IO banks (based on their IO standards), but they are incorrectly locked to non-HIGH_DENSITY IO banks. Please review and update the LOC constraints:
sfp0_tx_disable_b
CRITICAL WARNING: [DRC AVAL-326] Hard_block_must_have_LOC: The hard block IBUFDS_GTE4 cell ibufds_gte4_sfp_mgt_refclk_0_inst is missing a valid LOC constraint for placement assignment, normally supplied by IP generation or manually assigned using the LOC property. Unguided placement of this block may cause problems in routing or other issues. Please check your design and set a valid LOC for this block to avoid these problems.
INFO: [Vivado_Tcl 4-198] DRC finished with 2 Errors, 1 Critical Warnings
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-23] Error(s) found during DRC. Placer not run.
INFO: [Common 17-83] Releasing license: Implementation
67 Infos, 2 Warnings, 9 Critical Warnings and 3 Errors encountered.
place_design failed
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.
```

</details>

Complete error dump at https://gist.github.com/vmayoral/fe294556fe89ee18737ac706ee07c100

**Struggle 2**

If I bypass the previous error root cause, I bump into the following error which affects most of the definitions in the XDC:

<details><summary>:x: (summary) ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value</summary>

```bash
open_run: Time (s): cpu = 00:00:17 ; elapsed = 00:00:20 . Memory (MB): peak = 3434.082 ; gain = 728.406 ; free physical = 44205 ; free virtual = 62105
# write_bitstream -force fpga.runs/impl_1/fpga.bit
Command: write_bitstream -force fpga.runs/impl_1/fpga.bit
Attempting to get a license for feature 'Implementation' and/or device 'xczu9eg'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xczu9eg'
INFO: [Common 17-86] Your Implementation license expires in 15 day(s)
Running DRC as a precondition to command write_bitstream
INFO: [IP_Flow 19-1839] IP Catalog is up to date.
INFO: [DRC 23-27] Running DRC with 6 threads
ERROR: [DRC NSTD-1] Unspecified I/O Standard: 12 out of 18 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: led[7:0], clk_125mhz_n, clk_125mhz_p, reset, and sfp0_tx_disable_b.
[... cut ...]
ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
INFO: [Common 17-83] Releasing license: Implementation
7 Infos, 0 Warnings, 0 Critical Warnings and 3 Errors encountered.
write_bitstream failed
write_bitstream: Time (s): cpu = 00:00:08 ; elapsed = 00:00:06 . Memory (MB): peak = 3771.211 ; gain = 337.129 ; free physical = 44065 ; free virtual = 61965
ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors.
```

</details>

Complete error dump at https://gist.github.com/vmayoral/b31084a9d7f93812a24b19e98e62be44

alexforencich, 2023-03-06:
> ERROR: [DRC NSTD-1] Unspecified I/O Standard: 12 out of 18 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or [...]

Looks like maybe some of these signals should be removed from the top-level verilog file, and the ones that are necessary should get LOC and IOSTANDARD constraints.

> ERROR: [DRC BIVB-1] Bank IO standard Support: Bank 66 has incompatible IO(s) because: The LVCMOS33 I/O standard is not supported for banks of type High Performance.  Move the following ports or change their properties:
sfp0_tx_disable_b
> ERROR: [DRC PLHDIO-5] HDIO DRC Checks: The following IO terminals need to be placed in HIGH_DENSITY IO banks (based on their IO standards), but they are incorrectly locked to non-HIGH_DENSITY IO banks. Please review and update the LOC constraints:
sfp0_tx_disable_b

From looking at the schematic, this one makes no sense at all.  The pin is pulled up to +3v3, but it looks like it's wired to a pin with Vccio of 1.8v, without any level shifting?  That looks like a rather serious board design mistake, tbh.  But if Vccio is 1.8, then use LVCMOS18.  

> CRITICAL WARNING: [DRC AVAL-326] Hard_block_must_have_LOC: The hard block IBUFDS_GTE4 cell ibufds_gte4_sfp_mgt_refclk_0_inst is missing a valid LOC constraint for placement assignment, normally [...]

Looks like this didn't get any LOC constraints.

vmayoral, 2023-03-07:
Thanks for all of the above @alexforencich,

I re-iterated based on this (and the rest of our discussion, so thanks!) resulting in:

- ✅ LED pins, removed the extra ones in the HDL https://github.com/vmayoral/verilog-ethernet/commit/00c8abf087de0e39409d1124d9cb2da2ec647d7a
- ⚠️ `sfp0_tx_disable_b` pulled up to +3v3, but it looks like it's wired to a pin with Vccio of 1.8V without level shifting. Using `LVCMOS18` removed the error while building https://github.com/vmayoral/verilog-ethernet/commit/f2d3f071ea1a07b77070b29014d354d387c40761
- ✅ avoided using same pins to assign various things. Used the 25 MHz PL-connected clock and through a PLL generated the 125 MHz clock https://github.com/vmayoral/verilog-ethernet/commit/969ef0c08fc40a332bace990f1f95fdb4815de34
- ✅ Update IOSTANDARD of pins, Note the need to be mapping the K26 XDC, identify in there the corresponding VCCO and then, cross-reference that with the Datasheet of the KR260 board pin I/O https://github.com/vmayoral/verilog-ethernet/commit/af1f75834506805e62541237fa0079d2a8131fe1

Final branch available at https://github.com/vmayoral/verilog-ethernet/tree/kr260. My expectation was for things to build just fine this time but still bumping into the following two errors:

```
ERROR: [DRC NSTD-1] Unspecified I/O Standard: 4 out of 11 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: led[1:0], clk_25mhz_ref, and reset.

ERROR: [DRC UCIO-1] Unconstrained Logical Port: 6 out of 11 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: led[1:0], clk_25mhz_ref, reset, sfp_mgt_refclk_0_n, and sfp_mgt_refclk_0_p.
```

Further analyzing this:
- ❌ **[DRC NSTD-1] Unspecified I/O Standard**, typically indicates that a logical port in your design is not constrained to a physical location on the FPGA device. This is confusing since it's defined in the K26 SOM is [XDC](https://www.xilinx.com/member/forms/download/design-license.html?cid=29e0261a-9532-4a47-bb06-38c83bbbb8c0&filename=xtp685-kria-k26-som-xdc.zip). Pins affected:
  - led[1:0] 
  - clk_25mhz_ref, 
  - reset
- ❌ **[DRC UCIO-1] Unconstrained Logical Port**, similar to above, hints that a logical port in your design is not constrained to a physical location on the FPGA device
  - led[1:0]
  - clk_25mhz_ref
  - reset
  - sfp_mgt_refclk_0_n
  - sfp_mgt_refclk_0_p

To my uderstanding, the definitions in the `fgpa.xdc` file should be enough as they refer to the K26 SOM (which sets the `PACKAGE_PIN` for every one them)

vmayoral, 2023-03-08:
All right, managed to address the hurdles above after various iterations. One of the major issues being experienced is that though I had modified the part number in various places, I had overlooked the variable in the `Makefile`. So of course I was being driven nuts by pin conflicts. It was a different SoC after all. https://github.com/vmayoral/verilog-ethernet/commit/b084a8bf9e78d79e1b32c3d46e3bceacd41eea71 did it.

- I've also removed the reset signal from the XDC and hardcoded that in the PLL https://github.com/vmayoral/verilog-ethernet/commit/4902d98d924621633de162d1f6b55346a26695c5#diff-2a694dcf8aeb602499f5e2bbc7b57eb3a05fa94f540abe61080008d6f86d2079R242
- Removed IOSTANDARD definitions from SFP MGT REFCLK pins to mitigate CRITICAL WARNINGS https://github.com/vmayoral/verilog-ethernet/commit/3700157745fbcf4559d4e39dd12a5037825dccc7#diff-0bfc7823308e66729fd6b8005b41707e964822e4b2c28416d70456e2adc2732dR128-R129

Bitstream builds just fine. However, when programming the PL (testing for now without SD card, to avoid any *possible* interference, will scale from there once functional) and testing the link:

```
victor@accelerationrobotics:~/verilog-ethernet/example/KR260/fpga/fpga$ sudo ethtool enp3s0
Settings for enp3s0:
	Supported ports: [ FIBRE ]
	Supported link modes:   10000baseT/Full
	Supported pause frame use: Symmetric
	Supports auto-negotiation: No
	Supported FEC modes: Not reported
	Advertised link modes:  10000baseT/Full
	Advertised pause frame use: Symmetric
	Advertised auto-negotiation: No
	Advertised FEC modes: Not reported
	Speed: Unknown!
	Duplex: Unknown! (255)
	Auto-negotiation: off
	Port: Direct Attach Copper
	PHYAD: 0
	Transceiver: internal
	Supports Wake-on: d
	Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
	Link detected: no
```

This hints there's something wrong with the design, since I can see the link up perfectly fine with the ZCU102 reference design. No CRITICAL WARNING. Looking at the WARNING messages through the logs, the following caught my attention:

```
./fpga.runs/synth_1/fpga.vds:WARNING: [Synth 8-3917] design fpga has port sfp0_tx_disable_b driven by constant 1
```

however that doesn't seem to be the issue, as I can see the same being reported and shown in the ZCU102 design, which is known and confirmed to work:

| KR260 design | ZCU102 design |
|------|-------|
|[image] | [image] |

alexforencich, 2023-03-09:
Well, one minor thing that should be fixed, although it should make no difference when using a DAC: the TX disable connection on the ZCU102 is inverted (there is a transistor in line that provides the level translation) and as such has a _b in the name and is tied to 1.  But on the KR260, it's a direct connection, so the _b should be dropped and it should be tied to 0.  

Now, as for why it's not working, we'll have to work through a few things.  No link reported at the link partner definitely rules out a lot of possibilities.  Start with the pinout report and the clock summary from the timing report.  Make sure that all of the clock frequencies in the timing report are correct, and that everything is actually using the correct pins.

vmayoral, 2023-03-09:
Good news first, managed to get it working ✅ for the AMD `KR260`! There's no a new reference design example for this board. Submitted a cleaned PR for your review @alexforencich at  https://github.com/alexforencich/verilog-ethernet/pull/150.

As for the last struggles, It was indeed related to (thanks a lot!):

> ummmm, this is likely a problem:
>
> https://github.com/vmayoral/verilog-ethernet/blob/3700157745fbcf4559d4e39dd12a5037825dccc7/example/KR260/fpga/rtl/fpga.v#L242

After fixing that to a logic `0`, worked like a charm. Reference echo back design example works like a charm 👍 💥 .

Now, into the not so exciting news. While further testing the PL with the PS enabled I noticed that:
1. My board (KR260) doesn't seem to be able to boot the PS *normally* any further. Regardless of whether I power-boot it, or after having programmed known designs to work previously. The PS doesn't fully boot. I inspected the boot process through serial and after a few attempts obtained the following:

<details><summary>Boot prompt ❌ </summary>

```bash
Xilinx Zynq MP First Stage Boot Loader
Release 2022.1   Sep 16 2022  -  04:56:15
MultiBootOffset: 0x1F0
Reset Mode	:	System Reset
Platform: Silicon (4.0), Running on A53-0 (64-bit) Processor, Device Name: XCZUUNKNEG
QSPI 32 bit Boot Mode
FlashID=0x20 0xBB 0x20
Pr�NOTICE:  BL31: v2.6(release):0897efd
NOTICE:  BL31: Built : 04:58:29, Sep 16 2022

U-Boot 2022.01-g91ad7924-dirty (Sep 15 2022 - 23:00:49 -0600), Build: jenkins-BUILDS-2022.1-som_qspi_generation-131

CPU:   ZynqMP
Silicon: v3
Detected name: zynqmp-smk-k26-xcl2g-revA-sck-kr-g-revB
Model: ZynqMP SMK-K26 Rev1/B/A
Board: Xilinx ZynqMP
DRAM:  4 GiB
PMUFW:	v1.1
Xilinx I2C FRU format at nvmem0:
 Manufacturer Name: XILINX
 Product Name: SMK-K26-XCL2G
 Serial No: 50571A21CT3N
 Part Number: 5057-04
 File ID: 0x0
 Revision Number: A
Xilinx I2C FRU format at nvmem1:
 Manufacturer Name: XILINX
 Product Name: SCK-KR-G
 Serial No: 51011A21CT3N
 Part Number: 5100-01
 File ID: 0x0
 Revision Number: B
EL Level:	EL2
Chip ID:	xck26
NAND:  0 MiB
MMC:
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Bootmode: QSPI_MODE
Reset reason:	SOFT
Net:
ZYNQ GEM: ff0b0000, mdio bus ff0c0000, phyaddr 4, interface sgmii
eth0: ethernet@ff0b0000
ZYNQ GEM: ff0c0000, mdio bus ff0c0000, phyaddr 8, interface rgmii-id
, eth1: ethernet@ff0c0000
starting USB...
Bus usb@fe200000: Register 2000440 NbrPorts 2
Starting the controller
USB XHCI 1.00
Bus usb@fe300000: Register 2000440 NbrPorts 2
Starting the controller
[... 635 lines cut ...]
[  457.138169] Call trace:
[  457.140607]  dump_backtrace+0x0/0x1f0
[  457.144271]  show_stack+0x24/0x30
[  457.147577]  dump_stack_lvl+0x68/0x84
[  457.151240]  dump_stack+0x18/0x34
[  457.154547]  panic+0x144/0x354
[  457.157594]  do_exit+0x3dc/0x440
[  457.160814]  do_group_exit+0x44/0xac
[  457.164382]  get_signal+0x1e4/0x9b0
[  457.167862]  do_signal+0x1bc/0x2a0
[  457.171256]  do_notify_resume+0x164/0x2a0
[  457.175258]  el0_ia+0x114/0x120
[  457.178392]  el0t_64_sync_handler+0x124/0x130
[  457.182740]  el0t_64_sync+0x1a4/0x1a8
[  457.186396] SMP: stopping secondary CPUs
[  458.257026] SMP: failed to stop secondary CPUs 0,2
[  458.261807] Kernel Offset: disabled
[  458.265286] CPU features: 0x00002001,00000846
[  458.269635] Memory Limit: none
[  458.272686] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007 ]---
```

</details>

@giuliocorradi and @jasvinderkhurana, any thoughts in here? Have you guys seen or experienced something similar?

alexforencich, 2023-03-09:
Nice work!

That boot log seems to indicate that something may be wrong with the SD card?

vmayoral, 2023-03-10:
>  My board (KR260) doesn't seem to be able to boot the PS normally any further. Regardless of whether I power-boot it, or after having programmed known designs to work previously. The PS doesn't fully boot. I inspected the boot process through serial and after a few attempts obtained the following:

KR260's JTAG and microSD conflict when used together. Avoiding JTAG usage while booting did the trick.

----

Following from that, I flashed the FPGA from the PS manually. This gets the bitstream into the PL appropriately and the IP to operate as expected (✅ confirmed 10G reference example works while PS is active), **however**, it leaves the PS *unusable*. After programming the PL from the PS the following happens:

[image]

<details><summary>Errors observed in serial</summary>

```bash
kria login: [ 7579.771301] usb 1-1-port1: cannot reset (err = -71)
[ 7579.776314] usb 1-1-port1: cannot reset (err = -71)
[ 7579.781346] usb 1-1-port1: cannot reset (err = -71)
[ 7579.786530] usb 1-1-port1: cannot reset (err = -71)
[ 7579.791568] usb 1-1-port1: cannot reset (err = -71)
[ 7579.796467] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[ 7579.803030] usb 1-1-port1: cannot disable (err = -71)
[ 7579.808216] usb 1-1-port1: cannot reset (err = -71)
[ 7579.813245] usb 1-1-port1: cannot reset (err = -71)
[ 7579.818249] usb 1-1-port1: cannot reset (err = -71)
[ 7579.823235] usb 1-1-port1: cannot reset (err = -71)
[ 7579.828305] usb 1-1-port1: cannot reset (err = -71)
[ 7579.833206] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[ 7579.839780] usb 1-1-port1: cannot disable (err = -71)
[ 7579.844967] usb 1-1-port1: cannot reset (err = -71)
[ 7579.849981] usb 1-1-port1: cannot reset (err = -71)
[ 7579.854988] usb 1-1-port1: cannot reset (err = -71)
[ 7579.859994] usb 1-1-port1: cannot reset (err = -71)
[ 7579.865195] usb 1-1-port1: cannot reset (err = -71)
[ 7579.870144] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[ 7579.876731] usb 1-1-port1: cannot disable (err = -71)
[ 7579.881910] usb 1-1-port1: cannot reset (err = -71)
[ 7579.886935] usb 1-1-port1: cannot reset (err = -71)
[ 7579.891948] usb 1-1-port1: cannot reset (err = -71)
[ 7579.897057] usb 1-1-port1: cannot reset (err = -71)
[ 7579.902066] usb 1-1-port1: cannot reset (err = -71)
[ 7579.906966] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[ 7579.913541] usb 1-1-port1: cannot disable (err = -71)
[ 7579.918742] usb 1-1-port1: cannot disable (err = -71)
[ 7579.923960] hub 1-1:1.0: hub_ext_port_status failed (err = -71)
[ 7579.938249] usb usb1-port1: over-current condition
[ 7579.948638] usb usb3-port1: over-current condition
[ 7579.958241] blk_update_request: I/O error, dev sda, sector 31824944 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
[ 7579.969269] Buffer I/O error on device sda2, logical block 3715718
[ 7579.976611] blk_update_request: I/O error, dev sda, sector 12893592 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
[ 7579.987483] Buffer I/O error on device sda2, logical block 1349299
[ 7579.994376] blk_update_request: I/O error, dev sda, sector 90803024 op 0x1:(WRITE) flags 0x0 phys_seg 16 prio class 0
[ 7580.005124] Buffer I/O error on device sda2, logical block 11087978
[ 7580.011590] Buffer I/O error on device sda2, logical block 11087979
[ 7580.014415] blk_update_request: I/O error, dev sda, sector 90807808 op 0x1:(WRITE) flags 0x0 phys_seg 25 prio class 0
[ 7580.017919] Buffer I/O error on device sda2, logical block 11087980
[ 7580.034817] Buffer I/O error on device sda2, logical block 11087981
[ 7580.035013] blk_update_request: I/O error, dev sda, sector 28379512 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[ 7580.041147] Buffer I/O error on device sda2, logical block 11087982
[ 7580.057964] Buffer I/O error on device sda2, logical block 3285039
[ 7580.057966] Buffer I/O error on device sda2, logical block 11087984
[ 7580.057978] Buffer I/O error on device sda2, logical block 11087985
[ 7580.064291] blk_update_request: I/O error, dev sda, sector 68166400 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
[ 7580.087596] blk_update_request: I/O error, dev sda, sector 12892512 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[ 7580.099006] usb usb4-port1: over-current condition
[ 7580.103890] usb usb2-port1: over-current condition
[ 7580.114370] blk_update_request: I/O error, dev sda, sector 68166400 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 7580.170370] blk_update_request: I/O error, dev sda, sector 31824088 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[ 7580.181257] blk_update_request: I/O error, dev sda, sector 90769512 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[ 7580.201189] Buffer I/O error on dev sda2, logical block 563932, lost sync page write
[ 7580.209361] Aborting journal on device sda2-8.
[ 7580.209354] EXT4-fs error (device sda2): ext4_check_bdev_write_error:217: comm kworker/u8:3: Error while async write back metadata
[... 48 lines cut ...]
[ 7595.281154] EXT4-fs error (device sda2): __ext4_find_entry:1612: inode #59619: comm fancontrol: reading directory lblock 0
[ 7595.292304] EXT4-fs error (device sda2): __ext4_find_entry:1612: inode #59619: comm fancontrol: reading directory lblock 0
[ 7595.303394] EXT4-fs error (device sda2): __ext4_find_entry:1612: inode #59619: comm fancontrol: reading directory lblock 0
[ 7595.314711] EXT4-fs error (device sda2): __ext4_find_entry:1612: inode #59619: comm fancontrol: reading directory lblock 0
[... cut ...]
[ 7602.047713] systemd-journald[590]: Failed to write entry (14 items, 385 bytes), ignoring: Read-only file system
[ 7602.058698] systemd-journald[590]: Failed to write entry (14 items, 397 bytes), ignoring: Read-only file system
[ 7602.069756] systemd-journald[590]: Failed to write entry (14 items, 385 bytes), ignoring: Read-only file system
[ 7602.080577] systemd-journald[590]: Failed to write entry (14 items, 397 bytes), ignoring: Read-only file system
[ 7602.091403] systemd-journald[590]: Failed to write entry (14 items, 385 bytes), ignoring: Read-only file system
...
```

</details>

vmayoral, 2023-08-25:
Closing this ticket as the KR260 port was long ago validated. See https://github.com/alexforencich/verilog-ethernet/pull/150 for the upstream contribution attempt.

closed 2023-08-25 by vmayoral