alexforencich/verilog-ethernet #154: PTP

fengfpga, 2023-05-20:
May I ask how to add the PTP timestamp function in 10G Ethernet, I don't see the data adding port on the top layer.

alexforencich, 2023-05-21:
Only the MAC supports PTP timestamping at the moment, the rest of the UDP stack does not have provisions to pass the timestamps through.  I suppose that would not be too hard to add, at the expense of a lot of additional registers and BRAM bits to store the timestamps when enabled.  PTP protocol support is a different story, there are currently no plans to implement that.

fengfpga, 2023-05-21:
module ptp_clock #
(
    parameter PERIOD_NS_WIDTH = 4,
    parameter OFFSET_NS_WIDTH = 4,
    parameter DRIFT_NS_WIDTH = 4,
    parameter FNS_WIDTH = 16,
    parameter PERIOD_NS = 4'h6,
    parameter PERIOD_FNS = 16'h6666,
    parameter DRIFT_ENABLE = 1,
    parameter DRIFT_NS = 4'h0,
    parameter DRIFT_FNS = 16'h0002,
    parameter DRIFT_RATE = 16'h0005,
    parameter PIPELINE_OUTPUT = 0
)
(
    input  wire                       clk,
    input  wire                       rst,

    /*
     * Timestamp inputs for synchronization
     */
    input  wire [95:0]                input_ts_96,
    input  wire                       input_ts_96_valid,
    input  wire [63:0]                input_ts_64,
    input  wire                       input_ts_64_valid,

    /*
     * Period adjustment
     */
    input  wire [PERIOD_NS_WIDTH-1:0] input_period_ns,
    input  wire [FNS_WIDTH-1:0]       input_period_fns,
    input  wire                       input_period_valid,

    /*
     * Offset adjustment
     */
    input  wire [OFFSET_NS_WIDTH-1:0] input_adj_ns,
    input  wire [FNS_WIDTH-1:0]       input_adj_fns,
    input  wire [15:0]                input_adj_count,
    input  wire                       input_adj_valid,
    output wire                       input_adj_active,

    /*
     * Drift adjustment
     */
    input  wire [DRIFT_NS_WIDTH-1:0]  input_drift_ns,
    input  wire [FNS_WIDTH-1:0]       input_drift_fns,
    input  wire [15:0]                input_drift_rate,
    input  wire                       input_drift_valid,

    /*
     * Timestamp outputs
     */
    output wire [95:0]                output_ts_96,
    output wire [63:0]                output_ts_64,
    output wire                       output_ts_step,

    /*
     * PPS output
     */
    output wire                       output_pps
); I think this module may be the key module for generating timestamps, but I don't know how to obtain these input signals and their specific meanings. Is there any relevant documentation?

alexforencich, 2023-05-21:
Currently, there are basically 3 pieces implemented in this repo:

1. the PTP clock, which provides the time reference
2. a PTP CDC module, which is used to transfer the output of the PTP clock across clock domains
3. PTP timestamping support in the MAC modules

In general, you'll need one PTP clock module, which produces a 96-bit timestamp (output_ts_96), then this will drive the PTP time inputs on the MAC, likely via PTP CDC modules.  Then the MAC will return PTP timestamps for all packets, either on the TX PTP TS outputs for transmit frames, or in tuser for RX frames.  

I suggest taking a look at how it's wired up in Corundum.