alexforencich/verilog-ethernet #158: How can we apply zcu102 example project for all SFP port?

ahmetserdar, 2023-06-22:
Hi Alex,
I am able to make zcu102 example project work for sfp0 on petalinux. Now i want to make new project in order to make it work for all sfp. This is my full code what i did. [https://github.com/ahmetserdar/verilog-ethernet-zcu102/tree/main](url)
This project already works for sfp0, i can transmit data over sfp0 but when i try to make sfp1 work the sfp1_tx_tready signal goes zero.
All links are enable. What should i do for this issue?

Thank you.

ahmetserdar, 2023-08-02:
It works for sfp0 but when we plug in fiber cable to all sfp cage it doesn't work for even sfp0. What can be problem?

alexforencich, 2023-08-02:
this is the upstream example design doing this, or your modified version?

ahmetserdar, 2023-08-03:
We are working on petalinux project. This is my modified version. I only create multiple instantiation fpga core. I also modified your core for only sfp0 without creating multiple instantiation "https://github.com/ahmetserdar/sfp0_zcu102". Actually it works properly as i expect. When i try to make it work for all sfp there is a strange issue here. When i plug in fiber cable to sfp cage one by one (only one will be attached) , it works but at least two fiber cable are connected it doesn't work. I can not solve problem. Can you help to to figure out this issue?

ahmetserdar, 2023-08-03:
Hi Alex,

I solve problem. it is related to  jumper. It needs to connect J16,J17,J45 and one more jumper that is located over J45(there is no name)jumper to enable transmission over all sfp.

Thanks for reply..

ahmetserdar, 2023-08-04:
Hi Alex,

We want to transmit ADC data over 10G by using UDP protocol. There is no receiver case, only transmission. Do i need to write source and destination mac manually? Still there is strange issue, sometimes some sfp cage can not transmit any data but when i shutdown and power up again it works rarely. There is no deterministic issue.