alexforencich/verilog-ethernet #159: Create IP for the Alveo U280

ncppd, 2023-07-19:
Hello and thank you for this repository!
I am trying to understand how can I pack the Vivado Project into an IP, so I can use it with other projects.
Do you have any hints on how to do this?
Thanks!

eugene-tarassov, 2023-07-19:
I'm not sure you can use the whole RISC-V Vivado project with other projects. But you can re-use the generated RISC-V module in other other projects as is - it is just few pure Verilog and VHDL files with a simple and easy to use interface. I think you can package these Verilog/VHDL files as Vivado IP, but I have not tried this.

ncppd, 2023-07-20:
I am trying to port another project to the U280 (https://github.com/pulp-platform/snitch/tree/master/hw/system/occamy/fpga).
Since the original project uses the standard xilinx 1G ethernet IP and U280 has only QSFP ports, I am trying to figure out how can I use the 10G ethernet code in this repo as an IP in order to add it to the block design.

eugene-tarassov, 2023-07-20:
The ethernet controller itself is easy, it is one Verilog file - ethernet.v, plus a Linux driver.
But porting the code that connects the controller with QSFP port (the MAC part) is going to be an epic task.
You can use U250 board files in this repo as an example, I don't have files for U280.

Alternatively, you can try to use Xilinx proprietary IPs for QSFP ethernet.