alexforencich/verilog-ethernet #161: Will 100M/1G be supported?

shenkyw, 2023-08-17:
I'm highly interested in this project. However, I still need to connect my smartNIC to some legacy devices operating at 100M/1G speeds. It would greatly benefit me if this project could offer support for these 100M/1G connections.

alexforencich, 2023-08-17:
This repo already contains MACs for 10M/100M/1000M as well as 10G/25G.  Unless you meant to open this issue on the Corundum repository?

shenkyw, 2023-08-23:
Yes, exactly. I want to use Corundum with 100M/1000M support. I need low latency and those speeds for my setup. Currently, I'm using the Cisco Nexus SmartNIC K35-S. However, since newer low-latency network cards often don't support slower network speeds, I'm thinking that using Corundum on a newer card might be a solution.

alexforencich, 2023-08-23:
Well, low latency isn't really a design goal of Corundum, so if you need the lowest possible latency then you're probably going to be better off with a commercial solution.  But Corundum does run on the K35-S, so you should be able to test it out without too much difficulty.  

For 100M/1000M support, what exactly do you need?  1000BASE-X?  SGMII?  Something else?