alexforencich/verilog-ethernet #163: Modified example seems to not work

OroArmor, 2023-08-17:
I am trying to set up Ethernet on the [AC701 development board](https://www.xilinx.com/products/boards-and-kits/ek-a7-ac701-g.html) which uses the [Marvell 88E1116R](https://www.marvell.com/content/dam/marvell/en/public-collateral/transceivers/marvell-phys-transceivers-alaska-88e1116r-technical-product-brief-2011-05.pdf) PHY. To get the example to work on my FPGA, which is a variant of the XC7A200T, I modified the [Nexys Video](https://github.com/alexforencich/verilog-ethernet/tree/master/example/NexysVideo/fpga) by editing the constraints file. I also took some code from the [KC705](https://github.com/alexforencich/verilog-ethernet/tree/master/example/KC705/fpga_rgmii) example since the AC701 board has a 200MHz differential clock. 

As for the problems that I am having, the PHY LEDs seem to be active on just RX, and I never see a TX. The board does not show up using `arp -a` or when looking from a router's connected devices list. 

I was wondering if you had any advice on how to debug this issue or any knowledge about people having similar problems.

Thank you

alexforencich, 2023-08-18:
If RX works but not TX, it sounds like maybe there could be a delay issue on the TX data pins.  RGMII is DDR, hence getting the timing correct is less trivial than for an SDR interface where you can just invert the clock so that the data is updated on the falling edge.  So, either the PLL phase needs to be adjusted to something other than 90 degrees, or you might need to put in IODELAY blocks and tune the delay settings.  I should have a TCL script somewhere that lets you update the IODELAY tap settings directly in the bit file without rebuilding the whole project, so you can use that to experiment with the tap settings to find the middle of the window.

OroArmor, 2023-08-18:
I originally just copied the RTL files into a new vivado project, but after copying just the nexys video example and modifying that and then running `make`, I was able to get it to work with no additional changes. Thanks!

closed 2023-08-18 by OroArmor