alexforencich/verilog-ethernet #164: Use of eth_pcspma and SGMII/RGMII

russell-t, 2023-08-17:
Hello,

First off I just want to say thank you for making this resource available, it's a significant enabler for so many projects - thank you.

The issue I am having has to do with the eth_pcspma module. I am in the midst of creating a 1G RGMII example for the KR260, loosely following the discussion [here](https://github.com/alexforencich/verilog-ethernet/issues/10) where you suggested using the Nexys Video example as a starting point. In that example, an `eth_mac_1g_rgmii_fifo` module is instantiated in the fpga_core.v file.

However the VCU118 board uses the same PHY chip as the KR260, and in the VCU118/fpga_1g example, an `eth_mac_1g_fifo` is used in conjunction with a `gig_ethernet_pcs_pma_0` module, which seems to translate RGMII into SGMII which is what the PHY chip expects (?). The pcs/pma module seems to require a license and is fairly annoying to deal with based on other issue threads I've looked at.

Is there a way around using this module, or is it necessary to hook into the DP83867 PHY chip?

alexforencich, 2023-08-17:
Depends on how the PHY chip is wired up.  If it's wired for RGMII, then you should be able to use the RGMII wrapper.  If it's wired for SGMII, then at least as of right now you have to use the Xilinx PCS/PMA core.  We might make an open-source 1000BASE-X module at some point, but there is no timeline on that at the moment.  

For the KR260, I think the PL connected PHYs are all RGMII.

russell-t, 2023-08-18:
Thanks for the quick reply. Yes, looking at the VCU118 schematic I now see that the PHY is wired in such a way that only supports SGMII, whereas the PHY on the KR260 is wired to support RGMII, and that the PHY itself supports both RGMII and SGMII configurations. Thanks for clarifying that.

In the VCU118 example,`fpga.v` contains a a state machine on lines 362 - 487 that seems to set the PHY up in SGMII mode, I am guessing this will need to be changed appropriately to set it up in RGMII. It is unclear to me what the purpose of the `mdio_master` module instantiation immediately following that section is doing. Could you elaborate a bit on this?

alexforencich, 2023-08-18:
The state machine tells mdio_master what to do.  And I suspect that you might not need to poke anything via MDIO for the PHY chip to work for RGMII, so try it without the init state machine first.  IIRC the stuff that needs to be changed for the VCU118 design to work may be specific to SGMII.

russell-t, 2023-08-18:
Understood, thank you. After taking a closer look at the data sheet for the PHY it does appear that by default the chip is set up for RGMII. Just to clarify, are you suggesting that `mdio_master` may be unneeded entirely? Based on your description it sounds like without the state machine there would be no data for the module to send.

russell-t, 2023-08-18:
Documenting this for others who may see this. After creating a constraint for the 25MHz crystal oscillator and updating the MMCME2_BASE parameters accordingly, I am able to perform synthesis. However I am failing on implementation now with errors of SIM_DEVICE on cells `phy_rx_ctl_idelay` and `phy_rxd_idelay_x` not being specified and being set to ULTRASCALE_PLUS. It seems like maybe IDELAYE2 is not supported for Ultrascale+ and that I need to swap them out for IDELAYE3 modules. Does that seem right?

alexforencich, 2023-08-28:
Yeah, you'll need to use the correct primitives for the target device.  I guess that would be the `IDELAYE3` for US+.  I don't think I have an example that I can point you at for those, but presumably swapping those out should be pretty simple.

russell-t, 2023-08-28:
After some debugging I've got GEM2 "working" after swapping to IDELAYE3's and also to an MMCME4_BASE. The only thing is that the E3 instances work a bit differently than the E2s, there is now the option to specify number of taps ("count" mode) or number of picoseconds ("time" mode) with the DELAY_VALUE. My understanding is that in "count" mode temperature/voltage compensation isn't used and an IDELAYCTRL isn't needed, so the delay is not super accurate. However in "count" mode I am getting UDP packets, but the majority of them are being dropped. Working on getting "time" mode working at the moment.

russell-t, 2023-08-30:
Got it working! Only issue now is that sometimes on the initial loading of the FPGA overlay, it doesn't work entirely or takes a few seconds to start working. I think this may be an issue with the start up sequence. The data sheet of the PHY specifies that RESET_N must be pulled low for at least 1 microsecond on startup, and in the reference design this is tied to the inverted LOCKED signal of the MMCME. I don't think there is any guarantee the MMCM takes at least 1 microsecond to initialize.

sahasam, 2023-09-28:
Hey, I've been following the discussion trying to replicate russell's accomplishment on my KR260. I've created a new vivado project, brought in the necessary files by referencing the VCU118 Example Design, NexysVideo Example Design, and vmayoral's KR260 SFP design. I've got it synthesizing and generating a bitstream after converting to MMCME4_BASE and IDELAYE3 and dragging in constraints. However, the LED's on the connector still aren't turning on. I'd like to check a few assumptions here, if you wouldn't mind. For now, I'm only concerned about GEM2 PL.

1. No Poking via MDIO is necessary. I checked the strap resistors in the schematic and I see the MDIO address is 0x02, Autonegotiation is enabled and set to advertise 10/100/1000, SGMII is disabled, and both TX/RX clock skew is set to 0. Is this configuration really enough that no connection via MDIO is necessary?

2. No connection to the PS is necessary. I'm using xmutil to load my designs on the KR260 after scp'ing the bitfile and dtbo. As a result, I need to create a hardware platform which requires some boilerplate in the hardware design. I want to do everything in PL because that's just my intended use case, but I'm starting to suspect Xilinx doesn't like that idea, especially on the Zynq platform. I would imagine it's possible, but do you know anything about this?
[image]

3. The IDELAYE3's for the RGMII rxd and rx_ctl signals don't require any delays. I have them because the discussion says I need them, and the example design for the NexysVideo has them (my rgmii reference). [Here in the code](https://github.com/alexforencich/verilog-ethernet/blob/e84da8dbfb48689d41c9bd7b581721c1ebf5271c/example/NexysVideo/fpga/rtl/fpga.v#L232) the idelays just exist without really delaying anything. I carried that over to my design. This look okay to you?
```verilog
IDELAYCTRL #(
    .SIM_DEVICE("ULTRASCALE")
)
idelayctrl_inst
(
    .REFCLK(clk_500mhz_int),
    .RST(rst_125mhz_int),       // async reset
    .RDY()
);

IDELAYE3 #(
    .SIM_DEVICE("ULTRASCALE_PLUS"),
    .DELAY_TYPE("FIXED"),
    .DELAY_FORMAT("COUNT"),
    .REFCLK_FREQUENCY(500.0)
)
phy_rxd_idelay_0
(
    .IDATAIN(phy_rxd[0]),
    .DATAOUT(phy_rxd_delay[0]),
    .DATAIN(1'b0),
    .CLK(1'b0),
    .CE(1'b0),
    .INC(1'b0),
    .LOAD(1'b0),
    .CNTVALUEIN(8'd0),
    .CNTVALUEOUT(),
    .CASC_RETURN(1'b0),
    .CASC_IN(1'b0),
    .CASC_OUT(),
    .RST(1'b0),
    .EN_VTC(1'b0)
);
```
Please tell me I'm wrong somewhere, then maybe I can make some progress.

PS For those who found this through google, here's my pin constraints for PL GEM2 RGMII on the kr260 that I found by cross referencing the schematic with the XDC

```
set_property -dict {LOC E1 IOSTANDARD LVCMOS18} [get_ports RGMII_td[0]]  ;   # som_240_1_d7   HPA01P
set_property -dict {LOC D1 IOSTANDARD LVCMOS18} [get_ports RGMII_td[1]]  ;   # som_240_1_d8   HPA01N
set_property -dict {LOC F2 IOSTANDARD LVCMOS18} [get_ports RGMII_td[2]]  ;   # som_240_1_d4   HPA02P
set_property -dict {LOC E2 IOSTANDARD LVCMOS18} [get_ports RGMII_td[3]]  ;   # som_240_1_d5   HPA02N
set_property -dict {LOC F1 IOSTANDARD LVCMOS18} [get_ports RGMII_tx_ctl] ;   # som_240_1_c4   HPA00_CCN
set_property -dict {LOC A2 IOSTANDARD LVCMOS18} [get_ports RGMII_txc]    ;   # som_240_1_a3   HPA06P_CLK
set_property -dict {LOC A1 IOSTANDARD LVCMOS18} [get_ports RGMII_rd[0]]  ;   # som_240_1_a4   HPA06N
set_property -dict {LOC B3 IOSTANDARD LVCMOS18} [get_ports RGMII_rd[1]]  ;   # som_240_1_b7   HPA07P
set_property -dict {LOC A3 IOSTANDARD LVCMOS18} [get_ports RGMII_rd[2]]  ;   # som_240_1_b8   HPA07N
set_property -dict {LOC B4 IOSTANDARD LVCMOS18} [get_ports RGMII_rd[3]]  ;   # som_240_1_c9   HPA08P
set_property -dict {LOC A4 IOSTANDARD LVCMOS18} [get_ports RGMII_rx_ctl] ;   # som_240_1_c10  HPA08N
set_property -dict {LOC D4 IOSTANDARD LVCMOS18} [get_ports RGMII_rxc]    ;   # som_240_1_d10  HPA09P_CLK
```

russell-t, 2023-10-09:
Hi @sahasam, 

1. You need to add the reset_n pin to the PHY (LOC is B1) and ensure this pin is driven HIGH (I have it set as the inverted signal of the reset created by the MMCME4/sync_resets. Only after adding this was I able to see lights on the RJ45.
2. The MDIO... I am not sure. I have my MDIO state machine sending a software reset command (I believe it's 0x8000 to address 0x1F)
3. To get the IDELAYE3's working, this is the configuration I settled on:
```
.CASCADE("NONE"),
.REFCLK_FREQUENCY(300.0),
.DELAY_FORMAT("TIME"),
.IS_RST_INVERTED(1'b0),
.DELAY_TYPE("FIXED"),
.SIM_DEVICE("ULTRASCALE_PLUS"),
.DELAY_VALUE(240)
```
```
.CE(1'b0),
.DATAIN(1'b0),
.EN_VTC(1'b1),
.INC(1'b0),
.LOAD(1'b0),
.RST(1'b0)
```
`.IDATAIN` is connected to the input signal and `.DATAOUT` is the output signal. All other inputs/outputs not listed were left unconnected.

gabriserra, 2024-08-16:
Hi there. @russell-t, @sahasam did you manage to have a working setup at the end?
May I ask you if you are willing to share the source code, especially regarding the MDIO state machine?

Jayanth-20, 2025-10-22:
Hello all,
For implementing the GEM2 PL side ethernet on KR260 evaluation board I had took reference from this verilog-ethernet git.
As @russell-t commented in this issue I had made the changes for IDELAYE3's and MMCME4 by taking reference of @russell-t and by observing the vmayoral's KR260 SFP design (as it is nearer reference for PL ethernet). The bitstream had generated successfully I had programmed using hardware manager. I had entered the net netcat -u 192.168.1.128 1234. But there is no response and there is no LED's are blinking and I can't see any link establishment. So anyone can tell me the mistake?

Jayanth-20, 2025-10-25:
As taking refernce from  @vmaroyal for implementing KR260's SFG+ ethernet because it is on PL side of the KR260, I had updated the MMCME4_BASE block and IDELAYE3 block by taking the reference from this discussion which was described by @russell-t. And by cross verifying the KR260 schematic and xdc done with pin constraints, they are same as pin constraints provided by @sahasam. I had assigned the LOC B1 for "phy_reset_n" as it is the output of som and i/p for phy chip. But I had little bit of confusion, as these hints are not being able provide enough idea to implement it.
And a doubt that, bitfile generated by vivado is enough or should I make the Makefile??

[fpga.txt](https://github.com/user-attachments/files/23138267/fpga.txt)

Jayanth-20, 2025-10-27:
Another observations are:

1. RX_CLK should generate 125MHz when it is configured properly but it is generating 2.5MHz and 2.5MHz will be generated continuously without connection of RJ45 from the host.
2. I had tried to run the script files of netns and udp_test.py, the output log will be attached to this comment.
3. After loading the bitfile using fpgautil in petalinux, a warning is prompting like warning: usb1 port: overcurrent condition.
4. Is there any necessary include MDIO? because @gabriserra discussed about no poking of MDIO is necessary. If it is required how?
5. How to implement this ethernet resource for real world application rather than loopback? what changes are required?

[shell_test.txt](https://github.com/user-attachments/files/23157481/shell_test.txt)