alexforencich/verilog-ethernet #180: There is no data return in ZCU106 example design

JLFu, 2023-11-29:
Hi Alex,
     I'm tring to start the ZCU106 example design ,I have successfully compiled the project and programed it to my board, but when I tested it with the command netcat -u 192.168.1.128 1234, I found that no data was returned. When I test with hping 192.168.1.128 -2 -p 1234 -d 1024 command, 100% packet loss is returned. I didn't make any changes to the project. What could be the cause of this? Do you have more detailed testing steps here or other test methods?
    Thank you very much

AlexLao512, 2023-12-19:
Have you tried arping and looking in Wireshark?

JLFu, 2023-12-19:
I have tried arping and looking in Wireshark，after i use the' hping3 192.168.1.128 -2 -p 1234 -d 1024 command ', i can see the arp request was sent in wireshark ,but i can't capture the arp pakcet in vivado (i use ila to capture the axis interface between phy layer and mac layer)  ,and so there is no arp reply in wireshark . I also tried  to using the 10g/25g Ethernet Subsysterm instead of the phy layer in the example design of zcu106. This time ,i can capture the arp request packet in vivado, but still no arp reply packet was send by the fpga.figure1 is hte arp request packet i capture in vivado,
[image]
As conclusion ,after i using the 10g/25g Ethernet subsysterm i can receive the arp request packet but no arp reply packet was send.As a result, there are still problems