alexforencich/verilog-ethernet #185: Does this 10G ethernet library use pause frames for flow control?

myqlee, 2024-01-03:
Thank you for sharing this wonderful library. 
Does this 10G ethernet library use **_pause frames_** for flow control?

alexforencich, 2024-01-03:
Yes, the MAC recently gained the MAC control layer, and the AXI stream FIFOs gained depth status outputs. But, I don't think I have logic in the MAC + FIFO modules for generating pause frames automatically yet. So RX pause frames pausing TX will work (just need to set up the config signals correctly), but for TX I think you'll need to look at the RX FIFO depth and trigger pause frame generation appropriately. Don't forget that you'll probably want some sort of hysteresis.

myqlee, 2024-01-03:
But I didn't find the pause frame flag in the source code, such as destination address 0x0180c2000001, type 0x8808, opcode 0x0001.

alexforencich, 2024-01-03:
See https://github.com/alexforencich/verilog-ethernet/blob/master/rtl%2Feth_mac_10g.v#L163 . The opcodes and such are configurable, drive in the value you want.

myqlee, 2024-01-03:
oh, the version of the program I downloaded is not the latest. 
Thanks！