alexforencich/verilog-ethernet #187: VCU 128 Support!

ABDELRAHMAN123491, 2024-01-16:
I am trying to set up an ethernet connection with Xilinks VCU 128, however the board is not responding. So, does the code in this repo for 118 VCU is off-the-shelf applicable to VCU 128? Or I need to edit the code and if you could give some tips on the nuances in the modified code?

alexforencich, 2024-01-16:
Don't know, I don't have access to a VCU128.  Some modifications are likely required; check the schematic vs. the constraints file and top-level ports.

navidaadit, 2024-01-17:
@alexforencich we have updated the constraints and the related "TX/RX nibble" settings, however, whatever works for VCU118 doesn't seem to be working for VCU128. We got bit file without error but we couldn't ping. One difference I found is you had explicit MDC MDIO writes for VCU118. You set the autonegotiation timer (using CFG4 register) to be 11 ms from its default 2 us. However, the CFG4 register for VCU128 is readonly. [Here](https://www.ti.com/lit/ds/symlink/dp83867ir.pdf?ts=1705466413661&ref_url=https%253A%252F%252Fwww.google.com%252F) is the datasheet. Do you think this can be the issue? And any suggestions for workaround? Thanks a lot!

navidaadit, 2024-01-17:
Hi @alexforencich we have updated the constraint file from VCU118 to VCU128. There is no pin for PHY1_RESET_B  in the FPGA so we connected it to PHY1_PDWN_B_I_INT_B_O. 

We also had to change the nibble settings as follows (based on your note in the xilinx forum [here ](https://support.xilinx.com/s/question/0D52E00006iHmN0SAK/1g25g-pcspma-unroutable-internal-net-issues-on-vcu118-xvcu9p?language=en_US) and based on the comments [here](https://support.xilinx.com/s/question/0D52E00006hplVpSAI/vcu128-sgmii-core?language=en_US)) 

RxNibbleBitslice0Used: false
InstantiateBitslice0: true
TX_in_upper_nibble: true
TX Lane 0: DIFF PAIR 1
RX Lane 0: DIFF PAIR 2

It also created a dummy NC port which we LOC properly.

Since both VCU118 and VCU128 uses the same DP83867ISRGZ, I assume mdc mdio settings don't need to be changed. However, the board didn't ping. Do you have any suggestions what else may need to be changed? Thanks a lot for your time.

alexforencich, 2024-01-17:
Look at the SGMII core status vector, try forcing the link to other speeds, connect the ILA and look at the GMII signals.

Alternatively, if you donate a VCU128, I'll sort all of this out and add an example design for it.

jcrr92, 2025-08-05:
I’d like to help others with VCU128-related issues because I lost a full week of my life trying to understand the documentation and schematic of this board.

First of all, the PHY address indicated in the documentation is incorrect. For the VCU128, the strap configuration of the Texas Instruments PHY is different, so the actual address for this board is 1. With this strap configuration, you don’t need to send any command via MDIO to enable the SGMII clock. This is likely why many people on the Xilinx forums are seeing success using address 3—just by coincidence. 😄

On another note, you need to add the property "DIFF_TERM_ADV TERM_100" to the differential pair wires of the ref625 clock. I believe this detail is correctly handled by the Xilinx axi_ethernet IP when using the default configuration, but not when using SGMII mode. I’d hate to think it’s due to economic reasons on Xilinx's part.

To make things easier for others, I’ll attach my constraints file for the VCU128, a Tcl script for the SGMII IP, and the top file of my project. I’m sure that with this information, @alexforencich could add support for this board in the future.

Constraints

`
set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]
set_property BITSTREAM.CONFIG.SPI_FALL_EDGE YES [current_design]
set_property BITSTREAM.CONFIG.CONFIGRATE 63.8 [current_design]
set_property BITSTREAM.CONFIG.SPI_OPCODE 8'h6B [current_design]
set_property CONFIG_MODE SPIx4 [current_design]
set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
set_property BITSTREAM.CONFIG.UNUSEDPIN Pulldown [current_design]
set_property CONFIG_VOLTAGE 1.8 [current_design]

#############
# Clocks    #
#############
create_clock -name ethernet_clk_ibufg -period 10 [get_ports ethernet_clk_p]
create_generated_clock -name ethernet_clk_mmcm -source [get_pins u_xlnx_clk_wiz/inst/mmcme4_adv_inst/CLKIN1] -master_clock [get_clocks ethernet_clk_ibufg] [get_pins u_xlnx_clk_wiz/inst/mmcme4_adv_inst/CLKOUT0]

#############
# CDC       #
#############

set_false_path -from [get_clocks ethernet_clk_mmcm] -to [get_clocks IntTx_ClkOut1]
set_false_path -from [get_clocks IntTx_ClkOut1] -to [get_clocks ethernet_clk_mmcm]

########
# UART #
########

set_clock_groups -asynchronous -group [get_clocks uart_clk] -group [get_ports UART_RX]

############
# ETHERNET #
###########

create_clock -period 1.600 [get_ports refclk625_p]

set_false_path -to [get_ports { ext_mdio_i_0 ext_mdc_0}]
set_output_delay 0 [get_ports { ext_mdio_i_0 ext_mdc_0}]

## Sys clock
set_property -dict {PACKAGE_PIN BJ51 IOSTANDARD LVDS} [get_ports ddr4_clk_n]
set_property -dict {PACKAGE_PIN BH51 IOSTANDARD LVDS} [get_ports ddr4_clk_p]

## Ethernet clock
set_property -dict {PACKAGE_PIN BK3 IOSTANDARD LVDS} [get_ports ethernet_clk_n]
set_property -dict {PACKAGE_PIN BJ4 IOSTANDARD LVDS} [get_ports ethernet_clk_p]

## Reset
set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets RESET_IBUF_inst/O]
######################################################################
# Reset mapping
######################################################################
set_property -dict {PACKAGE_PIN BM29 IOSTANDARD LVCMOS12} [get_ports RESET]

## UART
######################################################################
# UART mapping
######################################################################
set_property -dict {PACKAGE_PIN BP26 IOSTANDARD LVCMOS18} [get_ports UART_RX]
set_property -dict {PACKAGE_PIN BN26 IOSTANDARD LVCMOS18} [get_ports UART_TX]

[... 429 lines cut ...]
        .s_axis_rdata_o         (m_axis_rxd_tdata),
        .s_axis_rkeep_o         (m_axis_rxd_tkeep),
        .s_axis_rvalid_o        (m_axis_rxd_tvalid),
        .s_axis_rready_i        (1'b1),
        .s_axis_rlast_o         (m_axis_rxd_tlast)
    );
     
    xlnx_ila u_xlnx_ila_dma_axi_master_stream (
        .clk       (ethernet_clk),
        .probe0    (ethernet_reset_i),
        .probe1    ({m_axis_rxd_tdata,m_axis_rxd_tkeep,m_axis_rxd_tvalid,m_axis_rxd_tready,m_axis_rxd_tlast}),//32
        .probe2    ('{default:0}),//5
        .probe3    ({mdc_o,mdio_i,mdio_o,mdio_t,sgmii_rst}),//32
        .probe4    ('{default:0}),//32
        .probe5    (lvds_clk),//1
        .probe6    ('{default:0}),//32
        .probe7    ({gmii_rxd_i,gmii_rx_en_i,gmii_rx_er_i,gmii_rxd_sync,gmii_rx_en_sync,gmii_rx_er_sync,gmii_txd_o,gmii_tx_en_o,gmii_tx_er_o,gmii_txd_sync,gmii_tx_en_sync,gmii_tx_er_sync}),//32
        .probe8    (lvds_rst),//1
        .probe9    ('{default:0}),//5
        .probe10   ({sgmii_clk_en_0, ext_mdc_0, ext_mdio_i_0,an_interrupt_0}),//32
        .probe11   (pcspma_status_vector)//64
    );
    
endmodule
`

alexforencich, 2025-08-05:
As this repo is deprecated, it will not get support for the VCU128 outside of some kind of extenuating circumstances.  I certainly could add an example design to Taxi, but either way someone will have to donate a VCU128, otherwise I cannot maintain it.