alexforencich/verilog-ethernet #188: ExaNIC_X10

SrodinW, 2024-01-16:
hi, i'm trying use this project for 1 sfp channel. And my phy-layer not get up. there is i have some quations about ip(gtwizard):
1. If i create ip, i should create only channel or common, or channel and common( i choose in structural options all include from core or smth should be from example?)
2. if i use only 1 sfp then COUNT in eth_xcvr_phy_quad_wrapper.v should be 1
3. why in your project in xdc mgt_refclk 161,13 but in verilog file fpga.v 156,25?
because  async gearbox 64|66b, then free-running clk should be 156, 25 or 125?

And you use 156,25 from ip, but i configurate it's from SI generator on i2c? i should do reset gt after i configurate clk from si ?

thank you for your work and help.

alexforencich, 2024-01-16:
The current X10 code should use tcl scripts to generate the GT cores.  The script generates two cores, one with the QPLLs for the first channel and one without for the additional channels.  For one SFP+, you'll need the "full" instance only, but the wrapper will figure that out automatically, just set the count to 1.  Make sure the ref clock freq matches between HW and the TCL scripts - either 156 or 161 works, so if you have a reconfigurable osc that defaults to 156, then just adjust the TCL script to 156 and make sure the clock constraints are 6.4 ns instead of 6.206.  The double check the clock summary in the timing report to make sure the frequencies are what you expect.  And yes if you do need to poke the clock generator, you'll want to make sure the whole transceiver gets reset, including the QPLLs.

SrodinW, 2024-01-17:
If i create GT core not from tcl, it will be different between your current project ? 
I thought that if i use ref clk as 156, that pcs works on 161, and so i shiuld generate and 156, and 161...

alexforencich, 2024-01-17:
You can create the core however you want.  The TCL just makes it Vivado version independent, makes it east to change some of the settings, and makes it easy to generate two cores that are identical save for the "common" block.

The ref clock in principle doesn't really matter.  The QPLLs multiply it up to half line rate, then the channel divides it down (tx/rx progdivclk).  You just need to make sure that whatever clock you use, the QPLL multiplies it up to the correct frequency.  Since the QPLLs have fractional dividers, you can run it from pretty much whatever ref clock you have - 161, 322, 644, 156, 312, 100, 125, etc.

SrodinW, 2024-01-17:
Fine. I got it 
thank you very much.