alexforencich/verilog-ethernet #194: How To Trigger ARP Mechanism?

yunusesergun, 2024-02-23:
Hello everyone and @alexforencich,

I have a question about ARP mechanism in @alexforencich ethernet code. I have searched on the code but I could not get answer because of code complexity.

I want to use the code but at the backside, I want ARP to be sent periodically even if ARP is done with destination end. As I understand from the code, when ARP is done, there is no ARP sent to end point.

How can I trigger ARP mechanism even if ARP is done and data are sent? Thanks in advance,

Regards,

alexforencich, 2024-02-23:
"trigger ARP mechanism" as in you want to send periodic broadcast "gratuitous ARP" packets?  I think I looked in to doing that a while ago, should be a relatively simple modification in arp.v.

yunusesergun, 2024-02-23:
@alexforencich Yes, I think I expressed the term wrongly. Do you have any suggestion for how and where to change? Or which part should I be aware of? That would be very helpful.