alexforencich/verilog-ethernet #196: Can this design be validated on Alveo u50 card ?

lizajoseph, 2024-02-26:
Hi,

I came across your design as I am looking for a Ethernet design with UDP listener to be implemented on FPGA. I have a Alveo u50 card installed in my machine. I was able to generate the bitstream of your design. How can I validate this design on the Alveo u50 card? I can see that there is a simulation testbench environment available using cocotb. But I want to validate the design on my target platform. Is there any test environment that is available to test on hardware?

Liza

alexforencich, 2024-02-26:
Just load it via JTAG.  For the U50, you'll need the special U50-specific JTAG adapter, I think the part number is HW-DMB-1-G.

lizajoseph, 2024-02-26:
I would like to test and validate the FPGA logic on the u50 card

lizajoseph, 2024-02-28:
I have programmed the bitfile on the Alveo u50 card via the u50 programming cable. Then I tried to test the u50 card using the udp_test.py, however I do not know the u50 card's IP address to run the udp_test.py. Could you let me know how to get the current IP address of the u50 card or how to set a custom IP address on the u50 card?

alexforencich, 2024-02-28:
https://github.com/alexforencich/verilog-ethernet/blob/master/example/Alveo/fpga_25g/README.md

https://github.com/alexforencich/verilog-ethernet/blob/master/example/Alveo/fpga_25g/rtl/fpga_core.v#L224

lizajoseph, 2024-02-29:
Hi Alex,

Thanks for sharing the Readme and the IP address details that is configured on the FPGA card.
After programming the bitfile to the u50 card. I tried the netcat command with a text "hello" and then it is stuck, Similarly I see the same behavior with the hping command. How should I proceed further to debug this issue of not receiving the packets echoed back from the FPGA? Attached screenshot for your reference.

alexforencich, 2024-02-29:
if the link partner NIC is not detecting a link, then it sounds like something is wrong with the FPGA config that's being loaded on the U50.  Which design specifically did you build, and specifically what are you using as a link partner?  (For example, the 25G version "fpga" will not link-up at 10G, you'll need to build the design in the "fpga_10g" folder as that configures the transceivers for 10G).  But you also say that the 10G ethernet example design results in a link up - what is this design?  Is this a Xilinx example design?

JssdFish, 2025-08-20:
> Yes, you are right. I had built the fpga_AU50 which you pointed out that it is 25G version. Now I have built the fpga_AU50_10g folder design and programmed the bitfile on the u50 FPGA card and now the UDP packets are getting echoed back from the FPGA. Thanks for all your help.
> 
> Note: The 10G Ethernet IP example design I used to test the 10G NIC card link status is from below link https://docs.xilinx.com/r/en-US/Vitis-Tutorials-Hardware-Acceleration/Using-GT-Kernels-and-Ethernet-IPs-on-Alveo

May I ask you a few questions? I'm having a similar issue as yours, but even with a 10G version bitstream on the u50 FPGA, I still can't receive the UDP echo. I would sincerely appreciate your help.

lizajoseph, 2025-08-21:
> > > > Yes, you are right. I had built the fpga_AU50 which you pointed out that it is 25G version. Now I have built the fpga_AU50_10g folder design and programmed the bitfile on the u50 FPGA card and now the UDP packets are getting echoed back from the FPGA. Thanks for all your help.
> > > > Note: The 10G Ethernet IP example design I used to test the 10G NIC card link status is from below link https://docs.xilinx.com/r/en-US/Vitis-Tutorials-Hardware-Acceleration/Using-GT-Kernels-and-Ethernet-IPs-on-Alveo
> > > 
> > > 
> > > May I ask you a few questions? I'm having a similar issue as yours, but even with a 10G version bitstream on the u50 FPGA, I still can't receive the UDP echo. I would sincerely appreciate your help.
> > 
> > 
> > Check if your 10G link is detected and up using ethtool as seen in below screenshot
> > <img alt="Image" width="709" height="637" src="https://private-user-images.githubusercontent.com/151124269/480321290-9af3f966-e83b-4aa3-b9ea-db72a12ecac0.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NTY5NzEsIm5iZiI6MTc1NTc1NjY3MSwicGF0aCI6Ii8xNTExMjQyNjkvNDgwMzIxMjkwLTlhZjNmOTY2LWU4M2ItNGFhMy1iOWVhLWRiNzJhMTJlY2FjMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyMVQwNjExMTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNGU0ZGI1MjUyYzQ2YzhhZDYxZDU1ZDBjN2QwMTE0YTk4ZjhjOWU2ZDM2MzMzYmI5MTg0MDVmYjI1ZjcxZWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.BiwBmHPb0UeQAQ9kowXkAPQhkh1bWV0ZKk07Kk-qfys">
> 
> Thank you for your reply. I can see that the status is "link detected," and the corresponding interface shows a "RUNNING" state in ifconfig. I noticed that when I burned the 25G version, the "RUNNING" state was not present, but after reburning the 10G version, the "RUNNING" state appeared. I believe the connection should be established, and the FPGA burning process should be fine. I also configured the MAC address using the arp -s command. However, when I execute netcat -u 192.168.1.128 1234, it still hangs without any response. Below is a screenshot for reference. (link partner ip is 192.168.1.146)
> 
> <img alt="Image" width="549" height="460" src="https://private-user-images.githubusercontent.com/16873979/480351467-9ffe2ad5-0665-4f3f-a468-d7422b8f15fe.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTE0NjctOWZmZTJhZDUtMDY2NS00ZjNmLWE0NjgtZDc0MjJiOGYxNWZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyODgzZDZjYTY4ZGE1YzMxODU1YTI3NjdmM2NjYTBkZWNhYjQwNTA5MGZlYWNmYmVmOTg0YzA1ZmQ1YjM5NzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lPr4uJGkG8adzgXx7Y9qT811199PwVmX8hawYBvRUdQ"> <img alt="Image" width="745" height="59" src="https://private-user-images.githubusercontent.com/16873979/480351951-30d6b8f2-9316-4359-a28e-499f302a00fd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTE5NTEtMzBkNmI4ZjItOTMxNi00MzU5LWEyOGUtNDk5ZjMwMmEwMGZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmMzY3NzNkOTk5YTBlMjcwNDYzNGY3ODc4YmNlMWFiMDIzMTE3N2M4ZWMwYjI4NDA3Nzg2MDRiNjhmMzdiOGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PepPARyo040N4afx1GAeSEWVIPbqQl_-nvgsrYk_aYA"> <img alt="Image" width="334" height="66" src="https://private-user-images.githubusercontent.com/16873979/480352768-6bef20a5-e19e-47dc-80b4-3ab383a46805.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTI3NjgtNmJlZjIwYTUtZTE5ZS00N2RjLTgwYjQtM2FiMzgzYTQ2ODA1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiMDY0MTc3YmIxNmUzODA2YzMxNjUwYjM4ZjdjYmE4N2QxOWMwNDhmYTA4ODAzOTI2MGNlMjcxZmFlOWRhOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7LeGuL-8Fndj_M1G59FBtjNENEhUaXOTSM36OIVOjqs"> <img alt="Image" width="580" height="67" src="https://private-user-images.githubusercontent.com/16873979/480353854-b4300e97-a13e-4096-a202-54eeaa036578.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTM4NTQtYjQzMDBlOTctYTEzZS00MDk2LWEyMDItNTRlZWFhMDM2NTc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRjMjc0NGRjZmViMGZhY2U4Y2VkYzMyYWE4NDY5MDFjZGNhY2JkZDYxMjQ1YTMxYWUwYWE2ZTFiNWI2N2IxYWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.H3FlLhED2RpOdhje8yFeLmpMRXCs60S_GfrMrSKXNBA">

[image]

My physical setup is as seen in attached screenshot, I programmed the bitfile on the Alevo u50 FPGA card and was able to observe the echo back with the netcat command

JssdFish, 2025-08-21:
> > > > > Yes, you are right. I had built the fpga_AU50 which you pointed out that it is 25G version. Now I have built the fpga_AU50_10g folder design and programmed the bitfile on the u50 FPGA card and now the UDP packets are getting echoed back from the FPGA. Thanks for all your help.
> > > > > Note: The 10G Ethernet IP example design I used to test the 10G NIC card link status is from below link https://docs.xilinx.com/r/en-US/Vitis-Tutorials-Hardware-Acceleration/Using-GT-Kernels-and-Ethernet-IPs-on-Alveo
> > > > 
> > > > 
> > > > May I ask you a few questions? I'm having a similar issue as yours, but even with a 10G version bitstream on the u50 FPGA, I still can't receive the UDP echo. I would sincerely appreciate your help.
> > > 
> > > 
> > > Check if your 10G link is detected and up using ethtool as seen in below screenshot
> > > <img alt="Image" width="709" height="637" src="https://private-user-images.githubusercontent.com/151124269/480321290-9af3f966-e83b-4aa3-b9ea-db72a12ecac0.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NTY5NzEsIm5iZiI6MTc1NTc1NjY3MSwicGF0aCI6Ii8xNTExMjQyNjkvNDgwMzIxMjkwLTlhZjNmOTY2LWU4M2ItNGFhMy1iOWVhLWRiNzJhMTJlY2FjMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyMVQwNjExMTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNGU0ZGI1MjUyYzQ2YzhhZDYxZDU1ZDBjN2QwMTE0YTk4ZjhjOWU2ZDM2MzMzYmI5MTg0MDVmYjI1ZjcxZWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.BiwBmHPb0UeQAQ9kowXkAPQhkh1bWV0ZKk07Kk-qfys">
> > 
> > 
> > Thank you for your reply. I can see that the status is "link detected," and the corresponding interface shows a "RUNNING" state in ifconfig. I noticed that when I burned the 25G version, the "RUNNING" state was not present, but after reburning the 10G version, the "RUNNING" state appeared. I believe the connection should be established, and the FPGA burning process should be fine. I also configured the MAC address using the arp -s command. However, when I execute netcat -u 192.168.1.128 1234, it still hangs without any response. Below is a screenshot for reference. (link partner ip is 192.168.1.146)
> > <img alt="Image" width="549" height="460" src="https://private-user-images.githubusercontent.com/16873979/480351467-9ffe2ad5-0665-4f3f-a468-d7422b8f15fe.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTE0NjctOWZmZTJhZDUtMDY2NS00ZjNmLWE0NjgtZDc0MjJiOGYxNWZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyODgzZDZjYTY4ZGE1YzMxODU1YTI3NjdmM2NjYTBkZWNhYjQwNTA5MGZlYWNmYmVmOTg0YzA1ZmQ1YjM5NzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lPr4uJGkG8adzgXx7Y9qT811199PwVmX8hawYBvRUdQ"> <img alt="Image" width="745" height="59" src="https://private-user-images.githubusercontent.com/16873979/480351951-30d6b8f2-9316-4359-a28e-499f302a00fd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTE5NTEtMzBkNmI4ZjItOTMxNi00MzU5LWEyOGUtNDk5ZjMwMmEwMGZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmMzY3NzNkOTk5YTBlMjcwNDYzNGY3ODc4YmNlMWFiMDIzMTE3N2M4ZWMwYjI4NDA3Nzg2MDRiNjhmMzdiOGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PepPARyo040N4afx1GAeSEWVIPbqQl_-nvgsrYk_aYA"> <img alt="Image" width="334" height="66" src="https://private-user-images.githubusercontent.com/16873979/480352768-6bef20a5-e19e-47dc-80b4-3ab383a46805.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTI3NjgtNmJlZjIwYTUtZTE5ZS00N2RjLTgwYjQtM2FiMzgzYTQ2ODA1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThiMDY0MTc3YmIxNmUzODA2YzMxNjUwYjM4ZjdjYmE4N2QxOWMwNDhmYTA4ODAzOTI2MGNlMjcxZmFlOWRhOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7LeGuL-8Fndj_M1G59FBtjNENEhUaXOTSM36OIVOjqs"> <img alt="Image" width="580" height="67" src="https://private-user-images.githubusercontent.com/16873979/480353854-b4300e97-a13e-4096-a202-54eeaa036578.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3NjEyNTEsIm5iZiI6MTc1NTc2MDk1MSwicGF0aCI6Ii8xNjg3Mzk3OS80ODAzNTM4NTQtYjQzMDBlOTctYTEzZS00MDk2LWEyMDItNTRlZWFhMDM2NTc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDA3MjIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRjMjc0NGRjZmViMGZhY2U4Y2VkYzMyYWE4NDY5MDFjZGNhY2JkZDYx [... cut ...]
> My physical setup is as seen in attached screenshot, I programmed the bitfile on the Alevo u50 FPGA card and was able to observe the echo back with the netcat command

Yes, my physical setup is the same as yours. I noticed you mentioned that the No. 1 of the 4x10G QSFP splitter cable should be connected. I saw that I was using cable 3, which did show "link detected" but failed to achieve UDP echo. After I switched the connection to cable 1, the netcat command use was ok, and the UDP packets are now getting echoed back from the U50 FPGA. Thank you very much for your help.