alexforencich/verilog-ethernet #197: How to simulate and test the verilog-ethernet design?

lizajoseph, 2024-03-04:
I would like to run and simulate the verilog-ethernet design. I have installed cocotb, cocotb-test, however I am not clear on how to run the tests for this design and view the simulation results in waveforms.

alexforencich, 2024-03-04:
Run the tests by running "make" in the appropriate directory (each test case is in a folder with a makefile and a python script).  If you want to look at waveforms, run "make WAVES=1".  This should produce an fst file that can be opened in gtkwave.

lizajoseph, 2024-03-06:
I invoked the behavioral simulation after loading the fpga.xpr, However the simulation in the waveform does not generate any clock, the mmcm_locked remains "0" and the qsfp_tx_n[3:0], qsfp_tx_p[3:0], qsfp_rx_n[3:0], qsfp_rx_p[3:0] is in "z" state. Attached screenshot for reference

alexforencich, 2024-03-06:
The Alveo example design testbench is here: https://github.com/alexforencich/verilog-ethernet/tree/master/example/Alveo/fpga_25g/tb/fpga_core

lizajoseph, 2024-04-15:
> The Alveo example design testbench is here: https://github.com/alexforencich/verilog-ethernet/tree/master/example/Alveo/fpga_25g/tb/fpga_core

Hi Alex,

When I try to run simulation using "make WAVES=1", I get the following error. Am I missing anything in the setup?

ERROR: results.xml was not written by the simulation!
make[1]: *** [/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:96: results.xml] Error 1
make[1]: Leaving directory '/home/liza.joseph@CORP.FLOWEDGE.IN/verilog-ethernet/example/Alveo/fpga_25g/tb/fpga_core'
make: *** [/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

lizajoseph, 2024-04-24:
The issue was with "scapy" that needs to be installed as most testcases uses scapy libraries. After installing scapy I was able to run the testcases