alexforencich/verilog-ethernet #202: How to download and install Icarus Verilog in linux environment?

lizajoseph, 2024-04-16:
I want to run the simulation tests for this design , however I am unable to run the test with the command "make WAVES=1". I get below error.

FST warning: ignoring signals in previously scanned scope test_fpga_core.core_inst.udp_payload_fifo.genblk9.
ERROR: results.xml was not written by the simulation!
make[1]: *** [/home/liza/anaconda3/lib/python3.9/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:96: results.xml] Error 1
make[1]: Leaving directory '/home/liza/verilog-ethernet/example/Alveo/fpga_25g/tb/fpga_core'
make: *** [/home/liza/anaconda3/lib/python3.9/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

I suspect that the Icarus Verilog simulator is not installed, however I do not know how to get this simulator installed. The link in this page  for Icarus Verilog does not exist anymore.

lizajoseph, 2024-04-24:
I was able to download and install icarus verilog version 10.3 from below link
https://sourceforge.net/projects/iverilog/files/iverilog/10.3/verilog-10.3.tar.gz/download

Kreijstal, 2024-12-03:
you can download iverilog with your package manager, no?