alexforencich/verilog-ethernet #203: `test_ip_eth_tx_64.py` hangs

EdwinEstep, 2024-04-19:
After the latest commit to `ip_eth_tx_64.v` (`9b5a8cf24aeeeee9d0eadabb3136f7e7722544e2`), the MyHDL testbench hangs indefinitely:
[image]

The only changes I made were to tell the testbench where to find the myhdl VPI bindings. Here's a sample of the log output:
[image]

Note that this doesn't happen before that commit. Also note that the 8-bit version does not hang, completing in about 16seconds on my machine:
[image]

lizajoseph, 2024-04-23:
> After the latest commit to `ip_eth_tx_64.v` (`9b5a8cf24aeeeee9d0eadabb3136f7e7722544e2`), the MyHDL testbench hangs indefinitely: [...]

Hi,

Could you let me know, how to run the simulation for this design. I have installed "cocotb", "cocotbext-axi, cocotbext-eth, and Icarus Verilog (version 10.3), however when I run make WAVES=1 in this path "verilog-ethernet/example/Alveo/fpga_25g/tb/fpga_core" I get following error

 /usr/local/bin/vvp -M /home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp -fst -fst
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1713866796
     0.00ns CRITICAL cocotb.regression                  Failed to import module test_fpga_core: No module named 'scapy'
     0.00ns INFO     cocotb.regression                  MODULE variable was "test_fpga_core"
     0.00ns INFO     cocotb.regression                  Traceback: 
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 203, in _discover_tests
                                                            module = _my_import(module_name)
                                                          File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 72, in _my_import
                                                            mod = __import__(name)
                                                          File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/_pytest/assertion/rewrite.py", line 168, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/liza.joseph@CORP.FLOWEDGE.IN/verilog-ethernet/example/Alveo/fpga_25g/tb/fpga_core/test_fpga_core.py", line 28, in <module>
                                                            from scapy.layers.l2 import Ether, ARP
                                                        ModuleNotFoundError: No module named 'scapy'
                                                        
Traceback (most recent call last):
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/__init__.py", line 233, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/__init__.py", line 316, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 174, in from_discovery
    return cls(dut, tests)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 153, in __init__
    for test in tests:
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 203, in _discover_tests
    module = _my_import(module_name)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/regression.py", line 72, in _my_import
    mod = __import__(name)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/_pytest/assertion/rewrite.py", line 168, in exec_module
    exec(co, module.__dict__)
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/verilog-ethernet/example/Alveo/fpga_25g/tb/fpga_core/test_fpga_core.py", line 28, in <module>
    from scapy.layers.l2 import Ether, ARP
ModuleNotFoundError: No module named 'scapy'
Traceback (most recent call last):
  File "/home/liza.joseph@CORP.FLOWEDGE.IN/anaconda3/lib/python3.9/site-packages/cocotb/__init__.py", line 332, in _sim_event
    scheduler.log.error(msg)
AttributeError: 'NoneType' object has no attribute 'log'
     0.00ns ERROR    gpi                                Passing event to upper layer failed
FST info: dumpfile test_fpga_core.fst opened for output.
FST warning: sim_build/cocotb_iverilog_dump.v:3: $dumpfile called after $dumpvars started,
                                                 using existing file (test_fpga_core.fst).
FST warning: ignoring signals in previously scanned scope test_fpga_core.

EdwinEstep, 2024-04-23:
It looks like you need to install `scapy`, which is used for constructing and deconstructing network packets for testing.

lizajoseph, 2024-04-24:
> It looks like you need to install `scapy`, which is used for constructing and deconstructing network packets for testing.

Thank you very much, I am able to run the simulation tests after installing scapy

EdwinEstep, 2024-04-24:
Glad to hear it!