alexforencich/verilog-ethernet #204: Jumbo Frame Support - Not Working

abu7770, 2024-04-24:
Hi First of all thank you for the flawless ethernet hdl, they work great. Currently I am using 1440 Bytes size for transmitting UDP packets but I need to send more than that say 8640 Bytes. i verified the code and changed the line     
parameter TX_FRAME_FIFO = 0,
parameter TX_DROP_OVERSIZE_FRAME = TX_FRAME_FIFO,
 in the file "eth_mac_10g_fifo.v"
which was default set to 1 to drop oversize frame if i am correct so i changed it to 0 to support higher byte count. but the fact is when i give higher byte count for single packet after this little change, the Tx tready signal is not going after it goes low in first packet itself(i saw that it initially goes low for ARP response and becomes high ). Do i need to change anything else to support higher byte count.
Thank you

fpgapsyc, 2024-04-24:
Change the FIFO size in the parameter and increase the UDP checksum calculation fifo size. If the UDP checksum fifo is full and the tlast doesn't come. The checksum calculation logic waits for the complete frame to come wgich never comes as the tready goes low after the fifo is full.

So i suggest you to change the fifo sizes to 9000 in both UDP checksum fifo as well as 10G fifo side for you to be able to do the transfer of more than 1440.

Also make sure, you use the eth hdr ready signal to maintain interframe gap for sending the next new frame.

alexforencich, 2024-04-24:
Yep, TX_FRAME_FIFO must be set, changing the FIFO sizes is the correct solution.

adnanakcan, 2025-07-17:
Dear Friends,
Is there anyone who can provide support for **Jumbo frames**? If so, how exactly should this be done? Can someone explain?