alexforencich/verilog-ethernet #205: SGMII design in Stratix 10 using Gx bank

fpgapsyc, 2024-04-24:
Hi Alex,

In the examples design, i saw your 10g implementation using the native transceiver. Could you please let me know how i can do a similar design for 1G ethernet using SGMII interface using a GX bank.

Thanks

alexforencich, 2024-04-24:
Unfortunately I don't have any soft IP for SGMII/1000BASE-X at the moment, so I recommend looking at Intel's soft IP.  Presumably they have something for a 1G PCS/PMA, but I'm not sure if it's free-of-charge like the Xilinx one.  I am looking at putting together soft IP for 1G to support dynamic switching between 1G and 10G as well as to support WR, but I have no timeline for that at the moment.

fpgapsyc, 2024-04-25:
I tried using the Intel Triple Speed Ethernet Ip but it only provides option for LVDS Serdes and not for GX bank. Can you help me how i can integrate the TBI interface with native transceiver IP.

I have configured the TSE as SGMII MAC and PCS. For the PMA what can i use?

fpgapsyc, 2024-04-29:
Hi Alex,

Waiting for your valuable suggestion on this. Please help me with the design generation.

alexforencich, 2024-04-29:
Buy, build, or contract?

Looks like https://www.intel.com/content/www/us/en/products/details/fpga/intellectual-property/interface-protocols/1g-10g.html or https://www.intel.com/content/www/us/en/partner/showcase/offering/a5b3b0000004cS2AAI/ethernet-pcs-1g25g.html would probably work.  There are probably more options.  Not sure about pricing on those - if you're affiliated with a university, you can probably get a license for the Intel 1G/10G core without too much trouble.

fpgapsyc, 2024-04-29:
i Would like to build

alexforencich, 2024-04-29:
Alright then, join the corundum zulip (https://corundum.zulipchat.com) and we can discuss in more detail.