alexforencich/verilog-ethernet #217: About  10g ethernet

LiShuang-codes, 2024-08-04:
I have a board with kintex7 part in it. What i can use is single gtwizard ip rather than gtwizard_ultrascale IP that most of the examples utilize. I would like to ask for advice on how to migrate to an IP core using GTX. For example, in
[example/ZCU106/fpga/rtl/eth_xcvr_phy_wrapper.v](this example), both [eth_xcvr_gt_full](eth_xcvr_gt_full) and  [eth_xcvr_gt_channel](eth_xcvr_gt_channel) use instantiated IP core of gtwizard_ultrascale.  I dont understand the parameters of them T_T.

alexforencich, 2024-08-05:
The current 10G PHY implementation requires the transceiver to support the asynchronous gearbox mode.  Unfortunately the 7 series does not support this mode.  My recommendation is to use the Xilinx 10G PCS/PMA instead, which has a no-charge license.  See the NetFPGA SUME design for an example of how to use this core.

LiShuang-codes, 2024-08-05:
Thank you for recommendation.

LiShuang-codes, 2024-08-07:
> The current 10G PHY implementation requires the transceiver to support the asynchronous gearbox mode. Unfortunately the 7 series does not support this mode. My recommendation is to use the Xilinx [...]

Hi, now i use  the NetFPGA SUME design for an example to develop a similar one and now, things got into some trouble. The connection with other computer established only less than a second(a flash, to be precise) and then disconnected immediately. And again a few seconds later. I delete something about  i2c,  si5324, sfp3 and sfp4 which i do not have these ports/parts. A dubious code is: 
In example/NetFPGA_SUME/fpga/rtl/fpga.v, line 379:
```verilog
sync_reset #(
    .N(4)
)
sync_reset_sfp_inst (
    .clk(sfp_coreclk),
    .rst(rst_125mhz_int || si5324_i2c_init_busy),
    .out(sfp_reset_in)
);
```
I change them to :
```verilog
sync_reset #(
    .N(4)
)
sync_reset_sfp_inst (
    .clk(sfp_coreclk),
    .rst(rst_125mhz_int), // " || si5324_i2c_init_busy" has been deleted because of no si5324 here.
    .out(sfp_reset_in)
);
```
I use ila to detect  necessary internal  signals, and these signals seems suspicious.
[image]
PHY connection seems properly established?
[image]
Am i Missing some signals? The ila file is here (please change .zip to .ila
[iladata.zip](https://github.com/user-attachments/files/16522508/iladata.zip)
): 
I have considered all possible solutions and this problem retains unresolved. God, plz help me...
 These are my fpga.v:
```verilog
/*
 
Copyright (c) 2014-2018 Alex Forencich

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.

*/

// Language: Verilog 2001

`resetall
`timescale 1ns / 1ps
`default_nettype none

/*
 * FPGA top-level module
 */
module fpga (
    /*
[... 360 lines cut ...]
    /*
     * Ethernet: SFP+
     */
    .sfp_1_tx_clk(sfp_1_tx_clk_int),
    .sfp_1_tx_rst(sfp_1_tx_rst_int),
    .sfp_1_txd(sfp_1_txd_int),
    .sfp_1_txc(sfp_1_txc_int),
    .sfp_1_rx_clk(sfp_1_rx_clk_int),
    .sfp_1_rx_rst(sfp_1_rx_rst_int),
    .sfp_1_rxd(sfp_1_rxd_int),
    .sfp_1_rxc(sfp_1_rxc_int),
    .sfp_2_tx_clk(sfp_2_tx_clk_int),
    .sfp_2_tx_rst(sfp_2_tx_rst_int),
    .sfp_2_txd(sfp_2_txd_int),
    .sfp_2_txc(sfp_2_txc_int),
    .sfp_2_rx_clk(sfp_2_rx_clk_int),
    .sfp_2_rx_rst(sfp_2_rx_rst_int),
    .sfp_2_rxd(sfp_2_rxd_int),
    .sfp_2_rxc(sfp_2_rxc_int)
);
endmodule

`resetall

```

LiShuang-codes, 2024-08-12:
Solved. The board is flawed. I changed a new one and it runs smoothly.

closed 2024-08-12 by LiShuang-codes