alexforencich/verilog-ethernet #222: Error  run make project for KC705 board

saman-coder, 2024-08-31:
I am using CYGWIN  for make fpga_rgmii  project in KC705 example. When I run the make file, I get the following error. Thank you for your help.
[image]

gabriserra, 2024-09-03:
Are you working on Windows? Each example has a folder `lib/eth` which is a symbol link towards the `rtl` folder in the root of the repository. If you cloned the repository on Windows, the symbol link probably does not work.

Try to create again the symbol link using the `mklink` command.

gabriserra, 2025-10-06:
> [@alexforencich](https://github.com/alexforencich) facing same issue in Ubuntu. Please guide me

@alishan73321, please read my comment.