alexforencich/verilog-ethernet #224: ADM-PCIE-9V3 example not working

harris-chris, 2024-09-27:
Hello

I'm trying to build the ADM-PCIE-9V3 example, and having some issues. I've done the following:

- Set `SUBDIRS = fpga_10g` in `fpga_25g/Makefile`, because I'd like to build the 10Gb/s version; beyond this there are no other changes to the repo
- Run `make`. No errors are reported and there doesn't seem to be anything unusual on the output. It looks like it all works fine.
- Directly connected the FPGA to another machine, on a 10Gb/s interface. Autonegotiation is turned off. The connection is a PLR4 40G SM QSFP+ transceiver on line 1 of a breakout cable connecting to a 10G SM LR transceiver.
- According to `ethtool`, the speed is correctly agreed as 10Gb/s, and the link is detected. 
- Tried to `hping 192.168.1.128 -p 1234 -d 1024` as the documentation suggests. There is no response.
- By looking at `tcpdump` whilst pinging, it looks like the FPGA is not responding to ARP `who-has` requests.
- If a static ARP entry for `192.168.1.128`/`02:00:00:00:00:00` was created in the connected machine, there was still no response to the `hping`.

Could I check that this all sounds correct, and there's nothing obviously wrong with my process above?

elliotpotts, 2024-10-02:
I have also been unable to get the ADM-PCIE-9V3 example working.

harris-chris, 2024-10-02:
Closing this because my problem was entirely unrelated to this repo. If anyone else has a problem with the ADM-PCIE-9V3 card, check the card's settings with the `avr2util` supplied by Alpha Data, it's possible to permanently alter the QSFP28 interface clock (which is what somebody had done to my card, and could then be re-set by the same utility).

closed 2024-10-02 by harris-chris

alexforencich, 2024-10-02:
Nice work sorting that out, never knew you could do that myself.  Although I suppose if you can change it back, it's not permanent, only nonvolatile.  Honestly with the fractional-N support in the QPLLs, there should be no reason to change the clock frequency, unless maybe you're doing something really strange.

elliotpotts, 2024-10-04:
To add, the utility can be downloaded and built from here:
https://support.alpha-data.com/pub/firmware/utilities/linux/
https://support.alpha-data.com/pub/firmware/utilities/windows/