alexforencich/verilog-ethernet #225: multiple UDP stream support

lizajoseph, 2024-09-27:
Can this design be scaled up to support multiple streams? I would like to use this design to consume 2 or more UDP streams, is it possible to update the existing design for this ?

cube1us, 2025-02-09:
Though I am not 100% positive, but I think it should work if you consume the UDP packets fast enough.  Say, for example, you want to match on two different ports.  So, when you match one or the other, you copy the FIFO to the appropriate consumer, and then go back and listen again.

My experience that leads to this conclusion was that when I was working with it when I was testing my application, I accidentally didn't consume (throw away) a bunch of UDP packets that preceded a needed ARP response packet to resolve a sender's IP address, and the UDP receive FIFO filled up and the ARP packet got lost.

barawn, 2025-03-04:
Yes, just use an axis_demux to split the outgoing UDP streams based on their port, and then use an axis_mux to combine any responses.

I personally use a bunch of Verilog preprocessor macros to help connecting AXI4-Streams so this might look a little confusing but the basic idea should make sense. 

https://github.com/barawn/exanic_turf_test/blob/master/hdl/udp_port_switch.sv