alexforencich/verilog-ethernet #229: export to ku060

andylgh, 2024-10-07:
I referenced the ExaNIC_X10 project and ported it to my own Ku060 board. Now, I have a problem where I downloaded the program to the board but the network card is not connected. The top-level  is as follows:

module udp_stack_10g #
(
    parameter                                   C_S_AXI_ADDR_WIDTH      = 8,
    parameter                                   C_S_AXI_DATA_WIDTH      = 32,
    
    parameter                                   TARGET                  = "XILINX",
    parameter                                   DW                      = 64,
    parameter                                   KW                      = DW/8,
    //  fpga
    parameter       [31:0]                      C_SRC_IP_ADDR           = {8'd192,8'd168,8'd2,8'd101},    
    parameter       [15:0]                      C_SRC_UDP_PORT          = 1234,   
    parameter       [31:0]                      C_SRC_GATEWAY_IP        = {8'd192,8'd168,8'd2,8'd1},   
    parameter       [31:0]                      C_SRC_SUBNET_MASK       = {8'd255,8'd255,8'd255,8'd0},  
    parameter       [47:0]                      C_SRC_MAC_ADDR          = 48'h00_0a_35_00_1e_53,        
    //  pc                                                
    parameter       [31:0]                      C_DST_IP_ADDR           = {8'd192,8'd168,8'd2,8'd100},
    parameter       [15:0]                      C_DST_UDP_PORT          = 4001  
)(
    //  AXI4_LITE                                                                                                                                                                         
    //  System Signals
    input   wire                                clk,                //  156.25Mhz
    input   wire                                reset_n,
    //  Slave Interface Write Address Ports
    input   wire    [C_S_AXI_ADDR_WIDTH-1:0]    s_axi_awaddr,
    input   wire                                s_axi_awvalid,                                                                                                                                                          
    output  wire                                s_axi_awready,

    //  Slave Interface Write Data Ports                                                                                                                                                                                
    input   wire    [C_S_AXI_DATA_WIDTH-1:0]    s_axi_wdata,
    input   wire    [C_S_AXI_DATA_WIDTH/8-1:0]  s_axi_wstrb,
    input   wire                                s_axi_wvalid,
    output  wire                                s_axi_wready,

    //  Slave Interface Write Response Ports
    output  wire    [1:0]                       s_axi_bresp,
    output  wire                                s_axi_bvalid,
    input   wire                                s_axi_bready,

    //  Slave Interface Read Address Ports                                                                                                                                                                              
    input   wire    [C_S_AXI_ADDR_WIDTH-1:0]    s_axi_araddr,
    input   wire                                s_axi_arvalid,
    output  wire                                s_axi_arready,

    //  Slave Interface Read Data Ports
    output  wire    [C_S_AXI_DATA_WIDTH-1:0]    s_axi_rdata,
    output  wire    [1:0]                       s_axi_rresp,
    output  wire                                s_axi_rvalid,
    input   wire                                s_axi_rready,
  
    //  user interface                          
    //  tx(fpga -> pc)                          
    input   wire    [DW-1:0]                    usr_s_axis_tdata,
    input   wire    [KW-1:0]                    usr_s_axis_tkeep,
    input   wire                                usr_s_axis_tvalid,
    output  wire                                usr_s_axis_tready,
    input   wire                                usr_s_axis_tlast,
    input   wire                                usr_s_axis_tuser,
                                                
    //  rx(pc -> fpga)                          
    output  wire    [DW-1:0]                    usr_m_axis_tdata,
    output  wire    [KW-1:0]                    usr_m_axis_tkeep,
    output  wire                                usr_m_axis_tvalid,
    input   wire                                usr_m_axis_tready,
    output  wire                                usr_m_axis_tlast,
    output  wire                                usr_m_axis_tuser,    
    
    
    //input   wire                                clk_100m_p,
[... 276 lines cut ...]
        
        // PHY connections
        .phy_tx_clk             (   xgmii_tx_clk_156mhz ),  //  o
        .phy_tx_rst             (   xgmii_tx_rst_156mhz ),  //  o
        .phy_xgmii_txd          (   xgmii_sfp_txd       ),  //  i    
        .phy_xgmii_txc          (   xgmii_sfp_txc       ),  //  i
        
        .phy_rx_clk             (   xgmii_rx_clk_156mhz ),  //  o
        .phy_rx_rst             (   xgmii_rx_rst_156mhz ),  //  o
        .phy_xgmii_rxd          (   xgmii_sfp_rxd       ),  //  o
        .phy_xgmii_rxc          (   xgmii_sfp_rxc       ),  //  o
        
        .phy_tx_bad_block       (                       ),
        .phy_rx_error_count     (                       ),
        .phy_rx_bad_block       (                       ),
        .phy_rx_sequence_error  (                       ),
        .phy_rx_block_lock      (   phy_rx_block_lock   ),
        .phy_rx_high_ber        (                       ),
        .phy_tx_prbs31_enable   (                       ),
        .phy_rx_prbs31_enable   (                       )
    );

endmodule

`resetall

andylgh, 2024-10-07:
i change the eth_xcvr_gt.tcl as follow：
set base_name {eth_xcvr_gt}

set preset {GTH-10GBASE-R}

set freerun_freq {125}
set line_rate {10.3125}
**set refclk_freq {156.25}**
set qpll_fracn [expr {int(fmod($line_rate*1000/2 / $refclk_freq, 1)*pow(2, 24))}]
set user_data_width {64}
set int_data_width {32}
set extra_ports [list {rxpolarity_in} {txpolarity_in}]
set extra_pll_ports [list {qpll0lock_out}]

set config [dict create]

dict set config TX_LINE_RATE $line_rate
dict set config TX_REFCLK_FREQUENCY $refclk_freq
dict set config TX_QPLL_FRACN_NUMERATOR $qpll_fracn
dict set config TX_USER_DATA_WIDTH $user_data_width
dict set config TX_INT_DATA_WIDTH $int_data_width
dict set config RX_LINE_RATE $line_rate
dict set config RX_REFCLK_FREQUENCY $refclk_freq
dict set config RX_QPLL_FRACN_NUMERATOR $qpll_fracn
dict set config RX_USER_DATA_WIDTH $user_data_width
dict set config RX_INT_DATA_WIDTH $int_data_width
dict set config ENABLE_OPTIONAL_PORTS $extra_ports
dict set config LOCATE_COMMON {CORE}
dict set config LOCATE_RESET_CONTROLLER {CORE}
dict set config LOCATE_TX_USER_CLOCKING {CORE}
dict set config LOCATE_RX_USER_CLOCKING {CORE}
dict set config LOCATE_USER_DATA_WIDTH_SIZING {CORE}
dict set config FREERUN_FREQUENCY $freerun_freq
dict set config DISABLE_LOC_XDC {1}

proc create_gtwizard_ip {name preset config} {
    create_ip -name gtwizard_ultrascale -vendor xilinx.com -library ip -module_name $name
    set ip [get_ips $name]
    set_property CONFIG.preset $preset $ip
    set config_list {}
    dict for {name value} $config {
        lappend config_list "CONFIG.${name}" $value
    }
    set_property -dict $config_list $ip
}

# variant with channel and common
dict set config ENABLE_OPTIONAL_PORTS [concat $extra_pll_ports $extra_ports]
dict set config LOCATE_COMMON {CORE}

create_gtwizard_ip "${base_name}_full" $preset $config

# variant with channel only
dict set config ENABLE_OPTIONAL_PORTS $extra_ports
dict set config LOCATE_COMMON {EXAMPLE_DESIGN}

create_gtwizard_ip "${base_name}_channel" $preset $config

andylgh, 2024-10-07:
xdc file as follow:

#100M
#set_property -dict {PACKAGE_PIN D23 IOSTANDARD LVCMOS12} [get_ports clk_100m]

#set_property -dict {PACKAGE_PIN E22 IOSTANDARD DIFF_SSTL12} [get_ports clk_100m_p]
#set_property -dict {PACKAGE_PIN E23 IOSTANDARD DIFF_SSTL12} [get_ports clk_100m_n]

set_property -dict {PACKAGE_PIN E22 IOSTANDARD LVDS} [get_ports clk_100m_p]
set_property -dict {PACKAGE_PIN E23 IOSTANDARD LVDS} [get_ports clk_100m_n]

set_property -dict {PACKAGE_PIN AP9 IOSTANDARD  LVCMOS33} [get_ports {leds[0]}]
set_property -dict {PACKAGE_PIN AN9 IOSTANDARD  LVCMOS33} [get_ports {leds[1]}]
set_property -dict {PACKAGE_PIN AP8 IOSTANDARD  LVCMOS33} [get_ports {leds[2]}]
set_property -dict {PACKAGE_PIN AN8 IOSTANDARD  LVCMOS33} [get_ports {leds[3]}]
set_property -dict {PACKAGE_PIN AL10 IOSTANDARD LVCMOS33} [get_ports {leds[4]}]
set_property -dict {PACKAGE_PIN AM10 IOSTANDARD LVCMOS33} [get_ports {leds[5]}]
set_property -dict {PACKAGE_PIN AE11 IOSTANDARD LVCMOS33} [get_ports {leds[6]}]

#REF_CLK
set_property -dict {PACKAGE_PIN J30} [get_ports sfp_a_refclk_n]
set_property -dict {PACKAGE_PIN J29} [get_ports sfp_a_refclk_p]

#SFP_A
set_property -dict {PACKAGE_PIN G34} [get_ports sfp_a_rx_n]
set_property -dict {PACKAGE_PIN G33} [get_ports sfp_a_rx_p]
set_property -dict {PACKAGE_PIN H32} [get_ports sfp_a_tx_n]
set_property -dict {PACKAGE_PIN H31} [get_ports sfp_a_tx_p]

set_property -dict {PACKAGE_PIN AG11 IOSTANDARD LVCMOS33} [get_ports {sfp_a_tx_dis}]
#set_property -dict {PACKAGE_PIN AH13 IOSTANDARD LVCMOS33} [get_ports {sfp_a_rs0}]
#set_property -dict {PACKAGE_PIN AJ13 IOSTANDARD LVCMOS33} [get_ports {sfp_a_rs1}]

set_property BITSTREAM.GENERAL.COMPRESS TRUE [current_design]
set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]

create_clock -period 6.400 -name sfp_a_refclk_p -waveform {0.000 3.200} [get_ports sfp_a_refclk_p]

# 161.1328125 MHz MGT reference clock
#create_clock -period 6.206 -name sfp_a_refclk_p [get_ports sfp_a_refclk_p]

alexforencich, 2024-10-07:
The rx_status signal is more reliable than rx_block_lock, so look at that one.  Do you see rx_status high when connected to an external 10G NIC?  Do you see rx_status high when you use a loopback SFP?

andylgh, 2024-10-08:
Does  rx_status refer to the following signal?

    output wire [6:0]             phy_rx_error_count,
    output wire                   phy_rx_bad_block,
    output wire                   phy_rx_sequence_error,
    output wire                   phy_rx_block_lock,
    output wire                   phy_rx_high_ber,

alexforencich, 2024-10-08:
By `rx_status` I mean `rx_status`: https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/eth_phy_10g.v#L81

andylgh, 2024-10-11:
I made modifications directly from the ExaNIC_X10 project, only modifying the pin constraints and ref_clk constraints. I found that the project also couldn't run ，and phy was not connected.The attachment contains the project that I have modified,
I can directly use the 10G/25G ETH subsystem of Xilinx system to see that phy can be connected. Attached is the project I modified. Can you give me some suggestions. vivado version： 2022.2
[ExaNIC_X10_ku060.zip](https://github.com/user-attachments/files/17344145/ExaNIC_X10_ku060.zip)