alexforencich/verilog-ethernet #245: Does this Ethernet project include Auto-Negotiation?

myqlee, 2025-01-23:
Thanks for this super nice project!
Does this Ethernet project include Auto-Negotiation? I looked at the RTL file name and there seems to be no such module.

alexforencich, 2025-01-23:
For Gigabit (SGMII and BASE-T), this is handled by the external PHY, so it is supported by this codebase (although for SGMII currently you need to use a vendor SGMII PCS/PMA core).  Support for at least 1000BASE-X is planned for the SV rewrite, I may also implement SGMII and AN as that is quite similar to 1000BASE-X but I don't have plans for that at the moment.  For 10G/25G, it's currently not supported, but I might look in to this for the SV rewrite.

myqlee, 2025-01-24:
Got it. Thank you for quick reply.