alexforencich/verilog-ethernet #250: GMII output DDR and input SDR

jalcim, 2025-02-25:
(In this line, if not "XILINX" and not "ALTERA" then ddr is automaticly activate...)[https://github.com/alexforencich/verilog-ethernet/blob/baac5f8d811d43853d59d69957975ead8bbed088/rtl/oddr.v#L135]

(called from ssio_sdr_out.v)[https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/ssio_sdr_out.v#L54]
with parameter IODDR_STYLE = "IODDR2"

(called from gmii_phy_if.v)[https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/gmii_phy_if.v#L101C18-L101C29]
with parameter IODDR_STYLE = "IODDR2"

(called from eth_mac_1g_gmii.v)[https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/eth_mac_1g_gmii.v#L185]
with parameter IODDR_STYLE = "IODDR2"

(but here on input its correctly sdr)[https://github.com/alexforencich/verilog-ethernet/blob/master/rtl/ssio_sdr_in.v#L162]

alexforencich, 2025-02-26:
ODDR is used to provide an inverted clock output that is guaranteed to be synchronous with the data since both the clock and the data will be driven from the same IOB flip flops.  The DDR output for RGMII does more or less the same thing, except it requires a 90 degree offset clock just for that ODDR flip flop to get the required alignment.

jalcim, 2025-02-26:
> ODDR is used to provide an inverted clock output that is guaranteed to be synchronous with the data since both the clock and the data will be driven from the same IOB flip flops. The DDR output for [...]

GMII uses an SDR clock aligned with the data. ODDR and the 90° phase shift are specific to RGMII for DDR transmission. They are unnecessary and irrelevant in GMII.

alexforencich, 2025-02-26:
At 1 Gbps, GMII needs 125 Mbps SDR data plus a 125 MHz full rate clock. The 125 MHz clock is effectively 0101 data transferred at 250 Mbps, hence the ODDR to generate the clock. If I used an SDR register, then either I would need to run the fabric clock at 250 MHz, or forward the clock directly without a register and have the tools figure out how to get the skew correct.

jalcim, 2025-02-26:
@alexforencich thanks for your instruction !

closed 2025-02-26 by jalcim

alexforencich, 2025-02-26:
Also it's worth remembering that GMII doesn't always run at 1 Gbps, it can down-rate to 10M or 100M MII, which runs at 2.5 (10M / 4) or 25 MHz (100M / 4) instead of 125 MHz (1G / 8). So simply forwarding the clock is doubly problematic as actually changing the internal clock frequency or switching between multiple clocks adds a lot of complexity; it's simpler to use an ODDR and a state machine to generate both single cycle clock enables for the MAC logic as well as use the ODDR to synthesize the lower-rate clock.