alexforencich/verilog-ethernet #261: From ZCU102 Porting to KC705

Keariu, 2025-08-08:
"I'm trying to port a design from a ZCU102 to a KC705. I need to replace the GTY IP with a GTX interface IP. I've made the change, but after connecting an Ethernet cable, the network shows as disconnected. What should I do?"
gtwizard_0  gtwizard_0_i
    (

    .sysclk_in(xcvr_ctrl_clk), // input wire sysclk_in
    .gt0_qplllock_out(xcvr_qpll0lock_out), // output wire gt0_qplllock_out
    .gt0_qpllrefclklost_out(), // output wire gt0_qpllrefclklost_out
    .gt0_qplloutclk_out(xcvr_qpll0clk_out), // output wire gt0_qplloutclk_out 
    .gt0_qplloutrefclk_out(xcvr_qpll0refclk_out), // output wire gt0_qplloutrefclk_out

     .soft_reset_tx_in(xcvr_ctrl_rst), // input wire soft_reset_tx_in
     .soft_reset_rx_in(xcvr_ctrl_rst), // input wire soft_reset_rx_in
     .dont_reset_on_data_error_in(1'b0), // input wire dont_reset_on_data_error_in
     .q1_clk0_gtrefclk_pad_n_in(sfp_mgt_refclk_0_n_ibuf), // input wire q1_clk0_gtrefclk_pad_n_in
     .q1_clk0_gtrefclk_pad_p_in(sfp_mgt_refclk_0_p_ibuf), // input wire q1_clk0_gtrefclk_pad_p_in

     .gt0_tx_mmcm_lock_out(), // output wire gt0_tx_mmcm_lock_out
     .gt0_rx_mmcm_lock_out(), // output wire gt0_rx_mmcm_lock_out
     .gt0_tx_fsm_reset_done_out(gt0_tx_fsm_reset_done_o), // output wire gt0_tx_fsm_reset_done_out
     .gt0_rx_fsm_reset_done_out(gt0_rx_fsm_reset_done_o), // output wire gt0_rx_fsm_reset_done_out
     .gt0_data_valid_in(1'b1), // input wire gt0_data_valid_in
 

     .gt0_gtxrxp_in                  (xcvr_rxp), // input wire gt0_gtxrxp_in
     .gt0_gtxrxn_in                  (xcvr_rxn), // input wire gt0_gtxrxn_in
     .gt0_gtxtxn_out                 (xcvr_txn), // output wire gt0_gtxtxn_out
     .gt0_gtxtxp_out                 (xcvr_txp), // output wire gt0_gtxtxp_out

    .gt0_txusrclk_out(), // output wire gt0_txusrclk_out
    .gt0_txusrclk2_out(), // output wire gt0_txusrclk2_out
    .gt0_rxusrclk_out(), // output wire gt0_rxusrclk_out
    .gt0_rxusrclk2_out(phy_rx_clk), // output wire gt0_rxusrclk2_out

    .gt0_rxdata_out                 (gt_rxdata), // output wire [63:0] gt0_rxdata_out
    .gt0_rxresetdone_out            (gt_reset_rx_done), // output wire gt0_rxresetdone_out
    .gt0_rxgearboxslip_in           (gt_rxgearboxslip), // input wire gt0_rxgearboxslip_in
    .gt0_rxdatavalid_out            (gt_rxdatavalid), // output wire gt0_rxdatavalid_out
    .gt0_rxheader_out               (gt_rxheader), // output wire [1:0] gt0_rxheader_out
    .gt0_rxheadervalid_out          (gt_rxheadervalid), // output wire gt0_rxheadervalid_out

    .gt0_txheader_in                (gt_txheader), // input wire [1:0] gt0_txheader_in
    .gt0_txdata_in                  (gt_txdata), // input wire [63:0] gt0_txdata_in
    .gt0_txresetdone_out            (gt_reset_tx_done), // output wire gt0_txresetdone_out

    //_________________________________________________________________________
    //GT0  (X0Y0)
    //____________________________CHANNEL PORTS________________________________
    //-------------------------- Channel - DRP Ports  --------------------------
        .gt0_drpaddr_in                 (1'b0), // input wire [8:0] gt0_drpaddr_in
        .gt0_drpdi_in                   (1'b0), // input wire [15:0] gt0_drpdi_in
        .gt0_drpdo_out                  (), // output wire [15:0] gt0_drpdo_out
        .gt0_drpen_in                   (1'b0), // input wire gt0_drpen_in
        .gt0_drprdy_out                 (), // output wire gt0_drprdy_out
        .gt0_drpwe_in                   (1'b0), // input wire gt0_drpwe_in
    //------------------------- Digital Monitor Ports --------------------------
        .gt0_dmonitorout_out            (), // output wire [7:0] gt0_dmonitorout_out
    //------------------- RX Initialization and Reset Ports --------------------
        .gt0_eyescanreset_in            (1'b0), // input wire gt0_eyescanreset_in
        .gt0_rxuserrdy_in               (1'b1), // input wire gt0_rxuserrdy_in
    //------------------------ RX Margin Analysis Ports ------------------------
        .gt0_eyescandataerror_out       (), // output wire gt0_eyescandataerror_out
        .gt0_eyescantrigger_in          (1'b0), // input wire gt0_eyescantrigger_in

    //----------------- Receive Ports - Pattern Checker Ports ------------------
        .gt0_rxprbserr_out              (), // output wire gt0_rxprbserr_out
        .gt0_rxprbssel_in               (0), // input wire [2:0] gt0_rxprbssel_in
    //----------------- Receive Ports - Pattern Checker ports ------------------
        .gt0_rxprbscntreset_in          (1'b0), // input wire gt0_rxprbscntreset_in
    //------------------------- Receive Ports - RX AFE -------------------------
        .gt0_rxdfelpmreset_in           (1'b0), // input wire gt0_rxdfelpmreset_in
        .gt0_rxmonitorout_out           (), // output wire [6:0] gt0_rxmonitorout_out
        .gt0_rxmonitorsel_in            (2'b00), // input wire [1:0] gt0_rxmonitorsel_in
    //------------- Receive Ports - RX Fabric Output Control Ports -------------
        .gt0_rxoutclkfabric_out         (), // output wire gt0_rxoutclkfabric_out

    //----------- Receive Ports - RX Initialization and Reset Ports ------------
        .gt0_gtrxreset_in               (1'b0), // input wire gt0_gtrxreset_in
        .gt0_rxpmareset_in              (1'b0), // input wire gt0_rxpmareset_in
    //---------------- Receive Ports - RX Margin Analysis ports ----------------
        //.gt0_rxlpmen_in                 (1'b0), // input wire gt0_rxlpmen_in
    //------------ Receive Ports -RX Initialization and Reset Ports ------------

    //------------------- TX Initialization and Reset Ports --------------------
        .gt0_gttxreset_in               (1'b0), // input wire gt0_gttxreset_in
        .gt0_txuserrdy_in               (1'b1), // input wire gt0_txuserrdy_in
    //---------------- Transmit Ports - Pattern Generator Ports ----------------
        .gt0_txprbsforceerr_in          (1'b0), // input wire gt0_txprbsforceerr_in
    //------------- Transmit Ports - TX Configurable Driver Ports --------------
        //.gt0_txmaincursor_in            (7'b0000000), // input wire [6:0] gt0_txmaincursor_in
    //---------------- Transmit Ports - TX Data Path interface -----------------

    //-------------- Transmit Ports - TX Driver and OOB signaling --------------
 
    //--------- Transmit Ports - TX Fabric Clock Output Control Ports ----------
        .gt0_txoutclkfabric_out         (), // output wire gt0_txoutclkfabric_out
        .gt0_txoutclkpcs_out            (), // output wire gt0_txoutclkpcs_out
    //------------------- Transmit Ports - TX Gearbox Ports --------------------

        .gt0_txsequence_in              (7'b0000000), // input wire [6:0] gt0_txsequence_in
    //----------- Transmit Ports - TX Initialization and Reset Ports -----------

    //---------------- Transmit Ports - pattern Generator Ports ----------------
        .gt0_txprbssel_in               (0) // input wire [2:0] gt0_txprbssel_in

);

Keariu, 2025-08-16:
> The PHY in this repo does not support the 7-series transceivers. I recommend using the Xilinx 10G PCS/PMA core instead.

Thank you for the suggestion. I've successfully ported the NetFPGA_SUME virtx7 series to the KC705. What are some ways to test its maximum speed using a 10-gigabit Ethernet card connected to the FPGA?

andreaabba, 2026-01-20:
Hi @Keariu I'm try to do your same thing of porting from NetFPGA_SUME  to zynq 7030 that is very similar to KC705. May you share your code? Thank You Andrea